OrcaSlicer/OrcaSlicer #8797: Tree Supports drastically different between 2.3.0 beta and 2.3.0 Beta 2

PaulWBush, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2

### Operating System (OS)

macOS

### OS Version

15.3.1 

### Additional system information

Apple M4 Pro
24 GB

### Printer

Bambu Lab A1

### How to reproduce

Slice the project file in 2.3.0 Beta and 2.3.0 Beta 2. Note the difference. The Beta 2 supports are much larger and cover much more of the model compared to the Beta supports. 

### Actual results

[image]
Sliced with 2.3.0 Beta 2

### Expected results

[image]
Sliced with 2.3.0 Beta

### Project file & Debug log uploads

https://drive.google.com/file/d/1hDs-tUb-IOfvkljccOd5A0YpcCQpSK04/view?usp=share_link

Project zip didn't want to upload. Maybe this works?

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I looked through the Beta 2 release notes and didn't see anything mentioned about changes to tree support generation. Hopefully this is an unexpected change?

Noisyfox, 2025-03-11:
This is due to the introdution of vertical support painting, now that painted support on the side will take effect.

PaulWBush, 2025-03-14:
Ooof. That's rough if you have to go back to old files and reslice them.