OrcaSlicer/OrcaSlicer #8800: Create Printer/Nozzle screen broken - Errors for both functions

vseven, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-beta2 (existed in beta 1 and 2.2.0)

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Printer

Creality Ender-3 V3 Plus

### How to reproduce

Select "Create Printer" in printer drop down
With "Create Printer" selected any vendor

Alternatively

Select "Create Nozzle from Existing Printer"

### Actual results

For creating a printer a error message stating "The model was not found; please reselect vendor"

For creating a nozzle the printer drop down is unavailable

### Expected results

The ability to select a printer and/or model

### Project file & Debug log uploads

No logs needed

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

Looks like a continuation of issue #4371 but that only lists the issue for the nozzle and it appears to be affecting creating a printer also.  That issue was closed due to no activity.  Guessing the "Model" drop down is not being populated.

vseven, 2025-03-10:
Fun side note:  In the latest version of BambuStudio, 1.10.3.76, on the same screen the "Create Printer" option does work (although the list of printers in there is not the same as what is available) but the "Create Nozzle for Existing Printer" option is also broken.

PuzzleParrot3D, 2025-03-11:
I'm experiencing these problems too. It looks like the nozzle creation side of the issue was also reported in issue #8677