OrcaSlicer/OrcaSlicer #8802: Creating a new Printer/Filament/Process from an existing user definition fails

tehniemer, 2025-03-10:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Linux

### OS Version

linuxserver/orcaslicer docker container

### Printer

User defined Voron Trident

### How to reproduce

Create a user defined profile for a printer, filament, or process, then create another user defined profile with some changes from the one just created and name it something different.

For example, create a custom profile for ABS filament, then create a new profile based on that for a different color.

Restart OrcaSlicer

### Actual results

The second profile created will revert to default values for the parameters common between the profiles, or it may not even show up in the list in some cases.

### Expected results

The profile should save correctly

### Project file & Debug log uploads

[logs.zip](https://github.com/user-attachments/files/19172066/logs.zip)

[project.zip](https://github.com/user-attachments/files/19172071/project.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

It seems that when saving a new profile from a user created profile the "inherits" field is populated with the user profile name and this breaks the new profile for some reason. To fix this I've had to go into comparison mode and uncheck everything but the inherits box and save the new profile again. 

[image]

Noisyfox, 2025-03-11:
Weird... normally when you create new profile from existing user profile the `inherits` field should set to the same value as the one you inherit from. I'll take a look.

Noisyfox, 2025-03-11:
By the way, can you tell me how did you create new profiles? Did you do that via this

[image]

Or by this

[image]

tehniemer, 2025-03-11:
I used the second method.

I migrated from SuperSlicer using this script https://github.com/theophile/SuperSlicer_to_Orca_scripts/ which may have been why the inheritance field was blank in the "original" profile. But that wouldn't explain why the new one wasn't also blank if the default is to copy the `inherits` field.

I know I've seen this question asked before, but I don't recall seeing an answer, what's the reason a user profile can't inherit from another user profile?

Noisyfox, 2025-03-11:
Because of the order of profile loading. It loads system profiles first then load customised one by one. And the exact order is unpredictable, which means it could load the 2nd customise profile before loading the 1st one, then it couldn't find its parent profile.

Noisyfox, 2025-03-12:
> But that wouldn't explain why the new one wasn't also blank if the default is to copy the `inherits` field.

That's expected because system profiles could have `inherits` field empty, and you want your custom profile to inherit from it. And for any non-system profiles this field should not be empty (unless you manually edited it, or generated from other programs, just like in your case), so this logic works (until it breaks).

DARKFiB3R, 2025-03-12:
This whole system is so confusing, always has been. I can never tell if it's not working, or I'm doing something wrong.

Is there a step by step guide anywhere, with idiot frendly explanations of the whys and wherefores?

Material settings, Custom Filaments, AMS Slots!!! None of them seem to be talking to each other right now.

The Create Filament options are confusing as hell too. I add a new vendor, choose pla, give it a random serial name, choose a preset (which I created in Material settings) .... Hold up!!!!  Turns out there are invisable boxes you can select next to the printers listed at the bottom! ffs.

OK, it's still not showing up in the ams section, and it's not listed when I go back to the custom filaments section. What tf is going on? I'm pulling my hair out over here 😭😭😭

 v2.3.0-rc

tehniemer, 2025-03-12:
> this logic works (until it breaks).

In any case it shouldn't fail silently like it is.

Maybe a solution would be to compare the name of the profile that will go into the `inherits` field against the database of system profiles and not create the inheritance if it's not found?