OrcaSlicer/OrcaSlicer #8816: Fill bed when print by object is selected

spydy98, 2025-03-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc build 79a5a9b

### Operating System (OS)

Linux

### OS Version

ubuntu 24.04/mint 22.1

### Printer

Ender 3 Pro .4mm 

### How to reproduce

1. Others
2. print sequence select by object
3. right click object 
4. fill bed with copies

### Actual results

fills the bed beyond capacity 

[image]

### Expected results

Should fill bed the same as print by layer 

[image]

### Project file & Debug log uploads

[Triforce Pick.3mf.zip](https://github.com/user-attachments/files/19194559/Triforce.Pick.3mf.zip)

[debug_Tue_Mar_11_09_43_32_2.log.0.zip](https://github.com/user-attachments/files/19194671/debug_Tue_Mar_11_09_43_32_2.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

charely6, 2025-04-03:
same issue, I think this use to work as expected?

charely6, 2025-04-03:
I think when "by object" is selected and it tries to run, it seems to have zeroed out the clearance around the object. because if you do fill the bed when its "by layer" it works as expected.

closed 2025-07-10 by github-actions[bot]