OrcaSlicer/OrcaSlicer #8817: Dragging layer slider switches to moving viewpoint if mouse leaves window

rgsteele, 2025-03-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

macOS

### OS Version

macOS 15.3.1

### Printer

Creality CR-6 SE

### How to reproduce

1. Slice a model
2. On the Preview tab, start dragging the layer slider
3. While dragging, move the mouse pointer off the application window

### Actual results

As soon as the mouse pointer leaves the application window, moving the mouse starts changing the view perspective rather than moving the layer slider.

### Expected results

The layer slider should continue to move as you move the mouse pointer, and the view perspective on the model should not change.

### Project file & Debug log uploads

[Test.3mf.zip](https://github.com/user-attachments/files/19195215/Test.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

evilDave, 2025-03-12:
I would love to see this fixed, it is very annoying. This has been logged (possibly several times) before. I had a go at fixing it, however I am just too unfamiliar with the gui library being used, and there is no commentary at all about how it is supposed to work.

jhenley01, 2025-06-26:
I agree this is very annoying and has been an issue for a long time.

pointful-dave, 2025-09-25:
Still a problem with 2.3.1-beta (though _someone_ seems to have worked on some ui issues, eg the line styles dropdown does seem to function a bit better without locking up under certain conditions - perhaps there was a gui lib update?)
Please do not close.

evilDave, 2025-09-25:
This is still an issue on 2.3.1-beta (although someone has done some ui work as the line type dropdown now functions much better without locking up under certain circumstances - maybe there was a gui lib update?)

Please do not auto-close.

iammattcoleman, 2026-10-30:
This still occurs on 2.4.2 (build 8500fcdc).

evilDave, 2026-10-30:
I can confirm it is fixed in the latest nightly. See #3744

evilDave, 2026-10-30:
The bug bot should probably suggest trying the latest nightly - as then you can see if something is fixed and just waiting in the pipeline to be released...

iammattcoleman, 2026-10-30:
> The bug bot should probably suggest trying the latest nightly - as then you can see if something is fixed and just waiting in the pipeline to be released...

Yup, sorry about that.

I can also confirm that it's fixed in the nightly build.