OrcaSlicer/OrcaSlicer #8818: Orca Slicer uses the relative coordinates without G92

60M6ER, 2025-03-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

OC: Windows 11

### Additional system information

When Orca Slicer changes extruder, it creates G1 with the relative coordinates.

[image]

In this case, G90 is used at the beginning of the file.

[image]

G91 is not in the g-code file.

### Printer

Custom FFF printer

### How to reproduce

1. Need enable Wipe tower.

It might be related to the fact that I have an offset between the extruders.

### Actual results

My printer is getting a critical error with negative coordinates.

### Expected results

G91 must be set before relative coordinates.

[image]

### Project file & Debug log uploads

[Project with error.zip](https://github.com/user-attachments/files/19195242/Project.with.error.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

SoftFever, 2025-03-13:
OrcaSlicer don't support G91 and won't generate relative coordinates.

In your project, the negative X coordinates is caused by the wrong extruder offset setting in your configuration.

60M6ER, 2025-03-14:
Does this mean I shouldn't set negative offset for extruders?

closed 2025-03-14 by 60M6ER

SoftFever, 2025-03-14:
It is machine-specific. For example, if it is a Klipper machine, you can set the minimum X position to -10; therefore, then -10 is a valid number.