OrcaSlicer/OrcaSlicer #8820: STEP is broken

streetgt, 2025-03-11:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 versios

### Operating System (OS)

Windows

### OS Version

Not relevant

### Printer

Not relevant

### How to reproduce

STEP import objects have changed

2.2.0
[image]

2.3.0
[image]

Please, fix this mess.

### Actual results

Import using 2.2.0 and 2.3.0 have changed.

### Expected results

Import should be exactly the same.

### Project file & Debug log uploads

Not relevant

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Noisyfox, 2025-03-15:
Will add an option like BambuStudio does.

streetgt, 2025-03-17:
> Will add an option like BambuStudio does in [bambulab/BambuStudio#5214](https://github.com/bambulab/BambuStudio/issues/5214) .

Awesome. Thank you. Its going to be really usefull.

PR #9102 (2025-04-05): Port STEP import dialog from BambuStudio
[image]

Thanks BambuLab!

Fix #8820
closed 2025-04-05 by SoftFever