OrcaSlicer/OrcaSlicer #8822: Wipe tower shell material

VIN-y, 2025-03-11:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

PETG-PLA multi material, wt tool-changer, wipe tower is unreliable at high z. That is because the two material does not stick well with each other. Thus, at higher z, the wipe is likely (often) fail. I can see some settings in the "Multimaterial" tab in Orca v2.3.0-RC to specify the specific material-to-feature; but, they don't seem to be applied to the wipe tower.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

The shell of the wipe tower should be given the option to be made of a single material (i.e. via one of the extruder).

### Describe alternatives you've considered

1. Like the current option in Prusa Slicer. There is a option to make to shell of wipe tower with a single material (i.e. extruder)
2. Also consider the option to exclude a nozzle (or nozzles). This way, only the PLA nozzle is excluded from the printing of the shell.

AliB10, 2025-03-25:
I’ve also found this an issue for prusa Xl 5t
In multilateral tab - filament for features if you set wipe tower to say extruder 4 it doesn’t change anything

This is extremely important for large wipe towers while printing pla / petg

kaantuncel, 2025-05-09:
Reiterating this - the Wipe Tower option seems to do nothing at all, even though the tooltip makes it clear it should perform similar to the PrusaSlicer option. 

This is a near mandatory enhancement for multimaterial set-ups, as most materials are not compatible enough to make the wipe tower stable.

ianalexis, 2026-09-23:
Fixed from a long time ago now.

closed 2026-09-23 by ianalexis