OrcaSlicer/OrcaSlicer #8829: Make Cubic the default infill

TimTheBig, 2025-03-12:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Cubic is a much superior infill, so I have to make a custom profile on every devise for every speed (ie. draft, standard, quality).

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Change the default infill to cubic.

### Describe alternatives you've considered

Gyroid is also a good option, however the extra were on printers is not worth it.

hotellonely, 2025-03-13:
> cubic在打印的时候,同一层会出现重合点,这会增加粘嘴,拉丝,堵头的风险,而Gyroid同层并无重合点,且强度和cubic其实在我看来很相似,差距并不大. 除非你需要cubic的内部填充分仓的特性,要不然保持Gyroid作为默认填充对新手友好.

cubic is much faster and creates uniform infill time. the chance of collision is also much lower than Grid, because the lines in cubic are shifted a little bit every layer, so the extra over extrusions on the inter-crossing spot are not going to layer on themselves and cause a bulge, which, is why Grid is infamous for being collision prone.

Gyroid while its generally a good option, it creates uneven layer times causing varying layer lines. Also the extra print time is quite high (often +50% time of cubic). It means that if you don't care about filament usage, you can essentially double the infill rate on cubic in the same printing time.

BelugaRex, 2025-03-20:
Cubic的填充对于物品的角支撑不好，特别是在填充密度低于15%的时候，很容易导致物品碎角，至于时间上，

7%填充，           总时间，总消耗
螺旋体，            6h44m，204.42g，47m49s
交叉层叠，        6h35m，205.06g，40m07s
立方体，            6h27m，202.45g，30m38s
自适应立方体， 6h22m，196.16g，23m36s

所以所谓的50%只是打印填充的时间，相对于总打印时间真的影响不够打，特别是对于pla材料我还会缓慢降温，从而强化成品强度，这个时间差距真的不太有必要，反而是对于模型的一些边角的保护我认为更为重要。
至于层打印时间，对数化之后真的不会在成品上有可见的差距，真正想要保持外观应该是设置外墙速度一致。

hotellonely, 2025-03-21:
> Cubic的填充对于物品的角支撑不好，特别是在填充密度低于15%的时候，很容易导致物品碎角，至于时间上，
> 
> 7%填充， 总耗时，总消耗，稀疏填充耗时 螺旋体， 6h44m，204.42g，47m49s 交叉层叠， 6h35m，205.06g，40m07s 立方体， 6h27m，202.45g，30m38s 自适应立方体， 6h22m，196.16g，23m36s
> 
> 所以所谓的50%只是打印填充的时间，相对于总打印时间真的影响不够大，特别是对于pla材料我还会缓慢降温，从而强化成品强度，这个时间差距真的不太有必要，反而是对于模型的一些边角的保护我认为更为重要。 至于层打印时间，对数化之后真的不会在成品上有可见的差距，真正想要保持外观应该是设置外墙速度一致。

a 7% sparse infill is definitely a rare case that has to be manually optimised for by the users themselves. gyroid isn't fool proof at this infill rate either. don't forget that gyroid can regress into a straight line sometimes.

for default settings, I'd say a 15% or even 20% cubic would be great.

BelugaRex, 2025-03-21:
我其实是从这个视频获取到的经验：https://www.youtube.com/watch?v=nV3GbN6hLjg&ab_channel=MadewithLayers%28ThomasSanladerer%29

所以我的模型基本上都是3层墙起步，即便是我为了我的健康在日常生活中主要的材料是透明的PETG，也很少见到层纹变化问题，
这也和我先前的回应中的观点一致，外墙的变化应该是由外墙速度去调整而非填充，

而我上述的例子用的7%填充，我日常打印其实蛮多7-10%的情况，相应的是5层的墙层，这能极大的提高物品的强度，所以我其实也认为默认墙应该是3而非是2。

当然我也在思考或许我的日常使用和大多数人应该不太一样，我是一个负责机器人团体的3d打印，所以很多时候会有很多奇妙的结构需要调整，而数年的经验和测试结果都是螺旋体大多数时候优于立方体，仅有少数的属于装饰件的，或者是大空腔的不需要受力的结构我选自适应立方体。

最后附赠一个尝试出来的符合你所说到的例子，一个orca自带的标准大小的兔子：

墙3顶5底5，填充15%，总耗时，总消耗，稀疏填充耗时
螺旋体， 4h07m，98.89g，1h29m
交叉层叠， 3h50m，99.75g，1h12m
立方体， 3h42m，100.33g，1h3m
自适应立方体， 3h24m，84.40g，46m3s

这个模型这个设置下确实好像是立方体，特别是自适应立方会很美好，但实际上这个模型上螺旋体不需要那么多的填充，

墙3顶5底5，螺旋体不同填充，总耗时，总消耗，稀疏填充耗时
15%， 4h07m，98.89g，1h29m
10%， 3h34m，82.37g，55m42s
7%， 3h16m，72.39g，37m9s

立方体并不适合低填充，因为线条之间的间距偏大，孔隙偏大，而螺旋体的空间均匀性比立方体更好，所以完全可以降低填充率，
所以我认为默认的设置应该是螺旋体，墙3顶5底5，填充10%，能有时间，强度时间与耗材的甜点值，

至于交叉层叠：
墙3顶5底5，交叉层叠不同填充，总耗时，总消耗，稀疏填充耗时
15%，3h50m，99.75g，1h12m
10%， 3h27m，83.06g，48m40s
7%， 3h13m，73.07g，34m58s

看上去好像美好，实际上在查看耳朵这种相对于身体属于小空腔的地方，孔隙有点过大，对打印的外观会有些许影响。

最后叠甲：我其实都用，无论是立方体，自适应立方体还是螺旋体，但作为默认设置我认为应该慎重再慎重，即便是说交叉点不多/偏移，也还是会增加堵头风险，对于很多觉得打印机是开箱即用的人们（虽然选择使用orca的人应该比较少这类人），能够减少烦恼是很好的事情。

hotellonely, 2025-03-24:
> 我其实是从这个视频获取到的经验：https://www.youtube.com/watch?v=nV3GbN6hLjg&ab_channel=MadewithLayers%28ThomasSanladerer%29
>
> 所以我的模型基本上都是3层墙起步，即便是我为了我的健康在日常生活中主要的材料是透明的PETG，也很少见到层纹变化问题， [...]

They chose Cross Hatch now, for some reason that you and I both don't understand :)

Adaptive Cubic is a total loss. It's highly unstable (because of its naive approach on calculating how to scale the infill intensity). 
In many cases Adaptive Cubic cannot reach a usable solution at all. It's the worst algorithm to be defaulted with. I know the word Adaptive sounds attempting, but no it would end up in tears in many cases. Even a high infill rate would not help, a 65% adaptive infill can use more material than a 99% adaptive infill, that's how bad it is.

I agree that Gyroid does better on giving a 3d support. However, we all know that the model's strength almost solely depend on the shells. And when Gyroid regress to a straight line, the support would be very poor for upper surfaces. This is especially bad when you have a low infill rate.

[image]
e.g, a top surface with this underneath is doomed to fail

[image]
however a top surface like this would be much better.
But they're both gyroid.

[image]
I would say that the only thing cubic needs is a higher infill rate. A 25% cubic gives you a nice and even support whenever you need it. The risk of Grid comes from self-crossing AND accumulated extrusion error. without accumulated extrusion error the risk of failing is extremely low for modern printers even the really bad ones.