OrcaSlicer/OrcaSlicer #8849: Crash on MacOS when resetting fillament profile

duke8253, 2025-03-12:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0rc

### Operating System (OS)

macOS

### OS Version

15.3.1

### Additional system information

M4 Mac mini, 24GB RAM

### Printer

Bambu Lab A1

### How to reproduce

1. Edit filament profiles settings of a custom filament, save it.
2. Open a project that uses the that custom filament profile.
3. Since the filament profile has been changed, it'll now show a `*` next to the filament.
4. Click edit filament, then click the reset icon, this causes a crash.

### Actual results

OrcaSlicer crashes immediately.

### Expected results

Resets the filament profile to the edited version.

### Project file & Debug log uploads

[crash_log.txt](https://github.com/user-attachments/files/19215713/crash_log.txt)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Hooba97, 2025-03-12:
same thing, it crashes when editing a printer as soon as clicking here 

[image]

duke8253, 2025-03-12:
> Screen.Recording.2025-03-12.155747.mp4 
> Cant reproduce it on my side. Am I doing it right? If so, you might want to include your config folder, and show which profile exactly is breaking the app

Ahh, I think I should be more specific, you'll need to click on the "reset" icon next to the name of the filament, not the reset for individual settings of the filament.

duke8253, 2025-03-12:
https://github.com/user-attachments/assets/ba32d277-7142-48e2-b9aa-1f84746a9d06

duke8253, 2025-03-12:
> > > Screen.Recording.2025-03-12.155747.mp4
> > > Cant reproduce it on my side. Am I doing it right? If so, you might want to include your config folder, and show which profile exactly is breaking the app
> > 
> > 
> > Ahh, I think I should be more specific, you'll need to click on the "reset" icon next to the name of the filament, not the reset for individual settings of the filament.
> 
>  Screen.Recording.2025-03-12.161634.mp4 
> Like this?

yep, not sure what's causing that for me, tested on two windows and one mac, all crashing for me.

duke8253, 2025-03-12:
Yeah, let me start a new thread on that other issue then, my profile has selected printers, supposed to be like this. 

[image]

Noisyfox, 2025-03-15:
Could you also give this PR a try and see if it solve the problem? https://github.com/SoftFever/OrcaSlicer/pull/8911

duke8253, 2025-03-15:
Still crashing for me, built it on my mac, haven't been able to setup my windows for building yet.

Noisyfox, 2025-03-15:
You could download the CI build. Thanks for the feedback, will look further.

duke8253, 2025-03-15:
Just confirmed still crashing on windows as well.

Noisyfox, 2025-03-15:
Just pushed a new commit to the PR that should fix the problem. Please give it a try.

PR #8911 (2025-03-16): Fix crash when resetting filament presets
Fix #8863
Fix #8849
closed 2025-03-16 by SoftFever