OrcaSlicer/OrcaSlicer #8856: 2.3.X 3DView Isn't centered in the selected plate anymore

Splarkszter, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 release candidate [Present in Previous 2.3.X versions]

### Operating System (OS)

Windows

### OS Version

Windoz 11

### Additional system information

irrelevant

### Printer

irrelevant

### How to reproduce

1. Open a new project (Or any project)
2. Orbit around
3. Add a new plate (or many)
4. Orbit around

### Actual results

Orbiting without a geometry selected (click on the plate or whatever, press ESC) with multiple plates in the file makes it spin around the original plate (plate 1) rather than the center of the selected plate.

### Expected results

Orbit around the selected/highlighted plate instead of the plate1 of the 3dView

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19223011/log.zip)

[Cube.zip](https://github.com/user-attachments/files/19223013/Cube.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

It happened at some point during 2.3.X releases. Prob starting with beta, it kept happening in beta2 too.

Noisyfox, 2025-03-15:
It's actually not spin around the original plate, but mostly because there are object on plate 1 that is close to the selected plate. Currently the distance threshold is almost 5x the plate size, and if any object is within that distance the camera will pick that object and orbit around it automatically.

Splarkszter, 2025-03-15:
I think it should be about being intuitive and easy to use rather than "make sense" or being "meaningless to".
My use case is to be able to easily inspect hole plates and alignment of those rather than having an offset orbit around an object.
I constantly work with multiple objects in plates and I often require precise alignments both on the place and object-to-object.

I hope this clip brings my issue in a more clear view of what's happening. It's annoying at best. (Also noticed that the click/drag-threshold to select an item it it's too low, my mouse slides by nothing and can't select anything. lol)

https://youtu.be/0eukxructDM

travisterrell, 2025-03-24:
> The logic behind it is that, usually it only make sense to orbit object. It's usually meaningless to look at an empty plate. Though I do think the current distance threshold is too large which could confuse user.

I would LOVE if we could get the default behavior to orbiting the object instead of the plate, like most other programs' interfaces do. I agree that especially in preview mode, it can be super difficult to zoom in and closely inspect a particular area, because orbiting even slightly moves the view to a completely different area! Then you have to zoom out, use middle or right click to move the view, zoom in again, repeat. (The only real workaround is to first move the object itself so that the particular portion of it that you want to inspect is directly in the center of the plate, but it requires moving it each time you want to inspect a different area, and even less ideal when you've carefully positioned things on your build plate for printing.)

_Of course, with 969 open issues—the majority of which are bugs—the odds of getting a new feature in place isn't great unless you do it yourself. Maybe one day._

Noisyfox, 2025-03-24:
> I would LOVE if we could get the default behavior to orbiting the object instead of the plate

That's already the current behavior, as long as the object is selected. Otherwise it will try find all objects that within the distance threshold of current plate, then orbit around the center of the objects.

Noisyfox, 2025-03-26:
> It's when you get over to Preview that it becomes an issue, which is where I tend to need to do the closest zoomed looking

Ah good point. I'll see if this could be possible.

PR #8912 (2025-04-05): Camera rotate around current plate center if plate is empty
Fix #8856
closed 2025-04-05 by SoftFever