OrcaSlicer/OrcaSlicer #8863: Crash issue in Multimaterial tab

GlauTechCo, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 Rc

### Operating System (OS)

macOS

### OS Version

Macos 15.3.2

### Additional system information

Mac mini M4  
Macbook air M1

### Printer

Flashforge 5mp

### How to reproduce

Crash video:

https://github.com/user-attachments/assets/b7b6ea9c-e154-4096-ae6b-e25b4d9e727b

### Actual results

None

### Expected results

None

### Project file & Debug log uploads

Crash video

### Checklist of files to include

- [ ] Log file
- [ ] Project file

duke8253, 2025-03-13:
seems related to #8849

Noisyfox, 2025-03-14:
Able to reproduce. Will try fixing it.

GlauTechCo, 2025-03-15:
@Noisyfox Hello, the problem is solved, thank you.

PR #8911 (2025-03-16): Fix crash when resetting filament presets
Fix #8863
Fix #8849
closed 2025-03-16 by SoftFever