OrcaSlicer/OrcaSlicer #8864: Space at the beginning of G-code START_PRINT

CodrutIftimie, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Linux

### OS Version

Linux Mint 22.1

### Printer

Creality K1C

### How to reproduce

Open a project
Slice
Generate G-code
Export G-code file
Open G-code file and search for START_PRINT

### Actual results

The M104  and START_PRINT lines begin with a space which make the printer not take the settings in consideration.

[image]

### Expected results

The lines start without a space.

[image]

### Project file & Debug log uploads

[obj_1_Paint Can Trigger.3mf.zip](https://github.com/user-attachments/files/19227283/obj_1_Paint.Can.Trigger.3mf.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

I noticed this happening on Linux. I have dualboot and on Windows (Orca 2.2.0) the specified lines do not begin with a space (that's how I was able to figure out the issue).
I think before I upgraded to 2.3.0 I still had this issue and was only able to notice it because I changed the filament type and saw the temps have different values.
On Linux i have installed the flatpak x86 version.

cochcoder, 2025-03-13:
@CodrutIftimie 
If you replace the start G-Code with the one below, does it fix your issue? The main and only difference is that a single space after `M104 S0` is removed. 
```
M140 S0
M104 S0
START_PRINT EXTRUDER_TEMP=[nozzle_temperature_initial_layer] BED_TEMP=[bed_temperature_initial_layer_single]
```

CodrutIftimie, 2025-03-13:
> [@CodrutIftimie](https://github.com/CodrutIftimie) If you replace the start G-Code with the one below, does it fix your issue? The main and only difference is that a single space after `M104 S0` is [...]

It doesn't because the values are [nozzle_temperature_initial_layer] and [bed_temperature_initial_layer_single]. If I change them to actual values it works.

The issue seems to be with the spaces at the beginning of the line, not at the end of it.

discip, 2025-03-13:
@CodrutIftimie
If you think this is the cause, why you don't just delete the spaces?

[image]

discip, 2025-03-13:
@CodrutIftimie
If you think this is the cause, why you don't just delete the spaces? 😊

[image]

CodrutIftimie, 2025-03-13:
@discip 
Because I never changed the G-code in printer settings and it looked just like in Windows so I never suspected that there are extra spaces there.
The profile is an import from the Windows export so I guess that's how it got there (I just checked in Windows and there is none).
@cochcoder
Now I see where you wanted me to put that G-code, I edited the sliced export and not the profile settings, my bad, I'm pretty new to this 😄.

Thank you both for the help.
Removing the leading space from Machine G-code section fixed the issue I encountered. I will close the issue.

closed 2025-03-13 by CodrutIftimie