OrcaSlicer/OrcaSlicer #8865: Bridge ends in the air for certain config

vovodroid, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Win10 x64

### Printer

Bambu A1

### How to reproduce

Set first layer 0.24, next 0.28

### Actual results

Bridge ends in the air

[image]

### Expected results

Bridge goes full gap

[image]

### Project file & Debug log uploads

[Bridge.zip](https://github.com/user-attachments/files/19227591/Bridge.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

discip, 2025-03-13:
There is nothing wrong with the slicer, your part is simply not parallel to the print bed. 😊
Just do this:

[image]

And your issue will be gone.

discip, 2025-03-13:
@vovodroid
There is nothing wrong with the slicer, your part is simply not parallel to the print bed. 😊
Just do this:

[image]

And your issue will be gone.

vovodroid, 2025-03-14:
Indeed, thanks!

closed 2025-03-14 by vovodroid