OrcaSlicer/OrcaSlicer #8873: Slicer fails to slice large file

FredsMedia, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Linux

### OS Version

Arch 

### Additional system information

CPU: 8 Core AMD Custom 0405 @ 3.50 GHz
GPU: AMD Custom GPU 0405 [Integrated] (roughly equivalent to a GTX 1050)
Memory: 14.48 GiB

### Printer

Ender-3 V3 KE

### How to reproduce

1 Import a detailed file (in my case it is a large lithophane)
2 Use a lithophane preset
3 Slice

### Actual results

It starts slicing and gets to around the infill stage before throwing an error (std::bad_array_new_length)

[image]

### Expected results

The file would slice just fine with no errors.

### Project file & Debug log uploads

I try but it just says: Failed to upload "log and project.zip" I will figure something out.

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

While slicing, orca never seems to use more than 2.6GiB of RAM despite the system having over 6GiB free.

FredsMedia, 2025-03-14:
Slicing the same model with the same machine and preset on my (weaker) Macbook works just fine and Orca uses about 3.2GB of ram while slicing.

Noisyfox, 2025-03-15:
Wait... Steamdeck?

FredsMedia, 2025-03-15:
Yep. Using CatchyOS (Modified Arch distro but I can actually use package managers)