OrcaSlicer/OrcaSlicer #8877: incorrect layer count in preview

zdravac, 2025-03-13:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

macOS

### OS Version

Sequoia 15.4 beta

### Additional system information

Mac Studio 2023 M2 Max
RAM 32 GB

### Printer

Bambu Lab X1C

### How to reproduce

1. open project file
2. slice & open preview
3. move layer slider on the right up to layer 868, move one layer up and it shows the correct layer, but the numbering is incorrect and it states 446, and when moving up it goes to layer 500 instead of 922 as it should. I am talking here about plate 3
4. same thing with plate 1 (from 868 + 1 goes to 446 and up to 1247). objetcts are just different in physical size hence different end layer number (and total nr of layers)

### Actual results

-

### Expected results

expected results would be correct numeration of layers

### Project file & Debug log uploads

[vases.3mf.zip](https://github.com/user-attachments/files/19236832/vases.3mf.zip)

[log.zip](https://github.com/user-attachments/files/19236943/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

screenshots of plate 3 sliced: 

[screenshots.zip](https://github.com/user-attachments/files/19236951/screenshots.zip)

Noisyfox, 2025-03-15:
According to your project settings, the correct layer count of plate 3 is 500, you have a 100mm high vase and 0.2mm layer height, which means 500 layers. And I couldn't reproduce the same error you described.
Though your screenshots do look odd. Could you also provide the gcode file so I could take a look?

zdravac, 2025-03-16:
> According to your project settings, the correct layer count of plate 3 is 500, you have a 100mm high vase and 0.2mm layer height, which means 500 layers. And I couldn't reproduce the same error you [...]

[Ebmeier_Vase_100mm non-spiral_PLA_2h30m.gcode.zip](https://github.com/user-attachments/files/19271516/Ebmeier_Vase_100mm.non-spiral_PLA_2h30m.gcode.zip)

closed 2025-06-23 by github-actions[bot]