OrcaSlicer/OrcaSlicer #8878: Incorrect Dependency of Dynamic PA: prediction model and Calibration

lookypanda, 2025-03-13:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

<html>
<body>
<p data-start="201" data-end="604">Earlier, I participated in testing the dynamic PA function during its development by user @igiannakas  (<a data-start="322" data-end="379" rel="noopener" target="_new" href="https://github.com/SoftFever/OrcaSlicer/pull/5609">link</a>). However, it appears that I misinterpreted my test results at that time. I have since redone the tests on a larger dataset using the PA tower method , which I found to be more precise and less influenced by external factors.</p>

<p data-start="606" data-end="940">I have analyzed the calibration data for Pressure Advance (PA) on my Ender 3 S1 with a 0.2mm nozzle. I hypothesize that PA is directly influenced only by layer height and acceleration. Since volumetric flow is the product of layer height, line width, and speed, changing the line width (and thus the flow) does not directly affect PA.</p>
<p data-start="942" data-end="973"><strong data-start="942" data-end="973">Observations from the Data:</strong></p>
<ul data-start="975" data-end="1733">
<li data-start="975" data-end="1236">
<p data-start="977" data-end="1236"><strong data-start="977" data-end="994">Layer Height:</strong><br data-start="994" data-end="997">
I observed that PA decreases as the layer height increases. For example, PA drops from 0.085 at 0.08 mm to 0.075 at 0.12 mm, and further to 0.06 at 0.18 mm. This suggests that a thinner layer requires a higher PA for proper compensation.</p>
</li>
<li data-start="1238" data-end="1455">
<p data-start="1240" data-end="1455"><strong data-start="1240" data-end="1257">Acceleration:</strong><br data-start="1257" data-end="1260">
At a fixed layer height of 0.12 mm, increasing the acceleration from 2000 mm/s² to 4000 mm/s² results in a decrease of PA from 0.075 to 0.06. This indicates that higher acceleration reduces PA.</p>
</li>
<li data-start="1457" data-end="1733">
<p data-start="1459" data-end="1733"><strong data-start="1459" data-end="1494">Line Width and Volumetric Flow:</strong><br data-start="1494" data-end="1497">
Even though changes in line width affect the volumetric flow, PA remains constant when layer height and acceleration are kept the same. This confirms that PA is not directly influenced by the flow but only indirectly via layer height.</p>
</li>
</ul>
<p data-start="1735" data-end="1842">I believe these observations support my hypothesis that PA depends solely on layer height and acceleration.</p>

Data Table:

[image]

Request:
I would like to discuss how this calibration hypothesis can be integrated into the firmware settings or the slicer configuration to optimize print quality. I believe that by adjusting PA based on layer height and acceleration, we can achieve more consistent and precise prints.

Thank you for your attention to this matter.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

Describe the solution I'd like:
I propose implementing a dynamic Pressure Advance (PA) calibration that automatically adjusts PA based solely on layer height and acceleration. In this solution, the slicer would calculate the optimal PA value using these two parameters—assigning a higher PA for thinner layers and lower acceleration, and vice versa—while disregarding direct adjustments based volumetric flow and acceleration.

That said, this new idea still requires further verification. I need to recheck and validate the dependency on other parameters, such as SCV, print speed, etc. Some factors are non-trivial; for example, the peak speed is influenced by acceleration—under low acceleration, the test object might not reach the peak speed, complicating the calibration. This means further testing is necessary to ensure that no subtle dependencies are overlooked.

Overall, by refining the model to focus on layer height and acceleration while re-evaluating the impact of additional parameters, I believe we can significantly improve print consistency and quality without the need for manual intervention.

igiannakas, 2025-03-14:
I know it’s probably much to ask but can you try a test where you vary the speed instead of line width to match the flow rates from the different layer heights please?

lookypanda, 2025-03-14:
> vary the speed instead of line width to match the flow rates from the different layer heights please

@igiannakas Yeah, no problem. I'm gonna try this test today. 

Also read yesterday's discussions in the original thread for this feature, and a couple of thoughts on this came up. I agree with parts of both positions. I'll try to post my thoughts today and test them later. 

p.s. just in case, I want to make it clear that I don't want to criticize the current implementation, but rather to help improve it and share my observations. The current implementation of the feature is useful even in case of inaccuracies, for example: when creating a profile for a certain layer height to use different accelerations and line widths (wide lines to print solid infill).

igiannakas, 2025-03-14:
No criticism at all taken - we are all learning.

I had experimented with different layer heights but initial conclusion was that the correlation was with flow as higher LHs produced higher flow in a similar way to higher speed. Line widths could vary the appearance of the test as they tend to squish the lines closer together.

But I’d love more data on this, hence the ask :) please try to match the resulting flow rates as closely as possible to make them comparative.

lookypanda, 2025-03-14:
@igiannakas 
**I ran two more tests. I have highlighted them in green on the screenshot.** 

Well, the results of the tests are controversial. In fact, we can see that PA has increased with decreasing speed. But at the same time it does not coincide with PA at the same flow at other speeds (and smaller layer).
I would also like to point out that with the slower speed(100mm/sec) the defects from incorrect PA are visually greatly reduced. 

[image]

### Hypothesis:
I believe that dynamic Pressure Advance (PA) is influenced by several factors, not solely by layer height and acceleration, as initially assumed. Please note that this is only a hypothesis based on my current testing, which I tried to evaluate without any preconceived expectations to avoid bias—especially since visual tests are inherently subjective and blind testing is challenging.

#### Layer Height:
I observed that layer height strongly influences PA. A thinner layer likely increases the back pressure of the extruded plastic due to the proximity of the build plate, which creates a narrower exit path for the filament. This might explain why lowering PA sometimes improves the appearance of bridges. However, it remains unclear why a similar dependency is not observed for line width. (But on the other hand the size of the ‘area’ of the gap for the plastic to exit from nozzle will not be to much affected by the line width when the extrusion starts.)

#### Acceleration and Speed:
It appears that both acceleration and printing speed significantly affect PA. Since PA is defined in terms of seconds, the duration of the acceleration phase (i.e., the time the printer head spends accelerating) plays a crucial role. At lower speeds or with higher acceleration, the printer reaches its target speed more quickly(in seconds), which seems to result in a lower required PA value(in seconds). This suggests that the interplay between speed and acceleration is a critical factor in determining PA. **Perhaps this complexity could be mitigated if PA were measured in the volumetric amount of extruded plastic rather than in time.**

#### High Acceleration or High Flow Rates:
Additionally, I hypothesize that at high accelerations or high flow rates, the extruder may operate more aggressively, leading to a faster pressure build-up and therefore a lower PA. However, the influence of layer height might compensate for this effect in some cases, resulting in unexpectedly and suspiciously consistent PA values even with variations in other parameters.

In summary, my hypothesis is that dynamic PA depends on a combination of layer height, acceleration, and speed — with potential secondary/inderect influences from factors such as flow rate. Further testing and analysis are required to validate these observations and fully understand the dependencies.

igiannakas, 2025-03-14:
Love this! Thank you. It’s similar to what I had seen with layer height also having an influence.

I’m away from home for a couple of weeks so can’t do any significant testing while away but this is something I’ll need to delve into when back.

Benik3, 2025-03-15:
That's pretty interesting and I think that I noticed it already (but didn't make such testing).
It looks like the first step should be rewrite the PA function in klipper, so it can automatically make corrections for such cases (or base it on another parameters, which will lead to correct calculations in all cases).

Henk72, 2025-03-17:
@lookypanda
I have the same experience with layer height having a huge impact in PA values. 

Should the hypothesis not be about the amount of squish. If you print with a high line width compared to layer height you will also get a lot of back pressure. E.g. 0.8 width and 0.2 height. Maybe not at extrusion start but for sure in continues lines and cornering.

lookypanda, 2025-03-21:
@igiannakas 
Hi. 
I wanted to ask you if it is possible to make a debug build/branch of the code where the current dynamic PA is changed a bit?
For example, for a test, try replacing flow with print speed and adding layer height. 

Or just add some additional parameters such as line width, height and speed( and have an optional option to specify 0 in unnecessary parameter columns) . 

So that it would be possible to test in printing the correctness of the PA dependence hypothesis. I would love to test it in practice.

igiannakas, 2025-03-22:
Yes of course; let me raise a PR

ShaneDelmore, 2025-03-29:
> That's pretty interesting and I think that I noticed it already (but didn't make such testing). It looks like the first step should be rewrite the PA function in klipper, so it can automatically [...]

The bleeding-edge-v2 branch of Kalico (formerly Danger-Klipper) has a couple of alternative non-linear pa implementations.  I have been testing it for about a week now and the tanh version has been quite good.  Worth checking out since you are interested in alternative klipper pa implementations.

igiannakas, 2025-03-29:
Is there a web page that explains these new options anywhere? Can’t find the v2 bleeding edge documentation for the life of me

lookypanda, 2025-03-29:
> Kalico (formerly Danger-Klipper) has a couple of alternative non-linear pa implementations
Can you please share a link to this? As I also couldn't find any mention of this in the documentation or pull requests, but it would be interesting to test.

lookypanda, 2025-03-29:
Oh, I think I found it. But it's not clear how it works or if there is any documentation.
https://github.com/KalicoCrew/kalico/pull/262

https://github.com/dmbutyugin/klipper/commits/advanced-features/ 
[image]

[image]

ShaneDelmore, 2025-03-30:
> Is there a web page that explains these new options anywhere? Can’t find the v2 bleeding edge documentation for the life of me

Docs aren't finalized yet but this worked well for me.  You can hit the Kalico discord to discuss as well.
https://github.com/CarVac/kalico/blob/main/docs/Nonlinear_Pressure_Advance.md

igiannakas, 2025-03-30:
Perfect, exactly what I was looking for! I’ll set it up and test :)!