OrcaSlicer/OrcaSlicer #8880: [x] Don't slow down outer walls - Not being honored

tlhintoq, 2025-03-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-RC

### Operating System (OS)

Windows

### OS Version

10x64

### Additional system information

[image]

### Printer

Bambu

### How to reproduce

Tick `[x] Don't slow down outer walls`
Slice

### Actual results

Outer walls are being slowed - even visible in gCode preview

[image]

### Expected results

Don't slow down for outer walls

### Project file & Debug log uploads

[TheAMOS.bambu.orca.zip](https://github.com/user-attachments/files/19238742/TheAMOS.bambu.orca.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

This does seem to be happening on a layer that also has support interface.

tlhintoq, 2025-03-14:
Closer look makes me think its something to do with how overhangs are being handled with respect to speed.

[image]

Its not the through-holes themselves because the bottom 75% of the holes isn't being slowed.
Its just the top portion of the holes where there is overhang.

As well as this supported part

[image]

Unticking `Slow down for overhangs [x]` helps some, but not all of it.  There's still 2 layers slowing down.

[image]

tlhintoq, 2025-03-14:
My personal feeling is the material setting should take precedent.  If that material has a requirement to keep it from having a visible defect as shown above, that's the material and transcends all print profiles: Draft, coarse, fine, custom - they still all require the material to maintain a constant speed on the outer face.

I shouldn't have to play with the profile settings just because I changed material - that's the point of having these values where they are.

Ataraxia-Mechanica, 2025-03-14:
This looks like extrusion rate smoothing jank to me, turning this off and only one wrong slowdown remain 

[image]

Ataraxia-Mechanica, 2025-03-14:
And the slowdown over the edge is caused by slow down for small perimeters.
[image]

SoftFever, 2025-03-15:
Thanks @Ataraxia-Mechanica 
It looks indeed the Extrusion rate smoothing caused this

tlhintoq, 2025-03-16:
Wow - looks like a lot of effort in to finding the several other settings that affect this.

So where does that leave this issue in the view of developers like @SoftFever ?

To me: I still hold to my idea that the print profile should be able to hold all these various choices.  And the material choice for `[x] Don't slow outer walls` should be considered when doing the calculations.
That way the profile for `Fine|Delecate` is what its supposed to be agnostic of material choice.  Then if a material needs the setting turned on, then its handled automatically by selecting that material.
Otherwise, the user finds themselves with a scratchpad on their desk listing the 5 settings they have to change every time they shift into a given material - which defeats the purpose of material settings and print profiles.  Nobody wants a spreadsheet of:
If I'm in `Fine|Delecate', and I choose Elegoo Rapid PETG, then I have to change this to on, that to off, this to 5, that to 10, turn off smoothing {...}
Seems it should be more like
`if (OUTER_WALL && DONT_SLOW_OUTER_WALL) == false {CALCULATE_SMOOTHING_SPEED}`
{conceptually though clearly not in the right coding language, but you get the gist, right?}

closed 2025-06-23 by github-actions[bot]