OrcaSlicer/OrcaSlicer #8884: Seam locking and/or additional options besides back

jhenley01, 2025-03-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

I often use the Back seam option and then rotate my model to get the seam where I want it to be. I find this often produces a cleaner result than seam painting, takes less time and it also means that I don't have to paint the seam again when I make a design change to the part and re-import it. 

I do a lot of batch printing and in order to fit the most parts on the plate I sometimes need to orient some of them in different directions, this means I lose the seam position on some of the parts.

In this example the Back seam setting puts the seam exactly where I want it, however I can only fit 10 instances on the plate this way.

[image]

If I rotate them 90 degrees I can fit two more instances on the plate but since they are in a different orientation the seam is no longer where I need it to be.

[image]

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

This could be solved in two different ways.

1. An option to "lock" the seam position once you have it where you want it. This way the seam would stay in the same place no matter how the part is oriented.

2. Offer additional seam position options besides Back. If there were additional settings like Front, Left, Right then they could be used as object modifiers. In the example above I could leave the 10 models set to back and the two rotated models to Left under per object settings to put the seam where I want.

hotellonely, 2025-03-14:
And an option to avoid putting seam on overhang corners.

tlhintoq, 2025-03-14:
Did you know...

You can paint the seam on the first object however/wherever you want it to be `on the part`.  Then when you copy and rotate etc. the seam stays the same on the copies?  I like to find a good spot on the interior side, preferably hidden in a crease of the object geometry.

See green painted seam in photo.

[image]

Zogar89, 2025-03-14:
I support the original request. When I posted my request, I received the same response. There's a big difference between what we asked for and the seam painting solution.

https://github.com/SoftFever/OrcaSlicer/issues/7799

Zogar89, 2025-03-26:
> I'd like to add that since most part cooling setups blow air from the
> front or sides the back is often the worst place to have the seam. You
> typically want the seam where you can get the best cooling, especially with
> overhangs.
>
> Please give us additional seam options besides back.

Great tip! Important point.