OrcaSlicer/OrcaSlicer #8887: Bottom Z distance not respected if support top Z distance is set to 0

Ataraxia-Mechanica, 2025-03-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 RC

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

### Printer

Voron 0

### How to reproduce

1. Set "Top Z distance" to 0, and "Bottom Z distance" to non 0
2. Observe there are no distance between support bottom interface layers and the object.

### Actual results

If "top Z distance" is set to 0mm, then the bottom Z distance on the generated support is also 0mm, regardless of what it has been set to.

[image]

### Expected results

[image]

### Project file & Debug log uploads

[X.zip](https://github.com/user-attachments/files/19244134/X.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

This makes supports without Independent support layer height unusable in cases where top Z distance should be 0 and bottom Z distance should not, as setting top Z distance to 0.001 creates a one layer high gap between supported layer and interface layers, but setting it to 0 makes both top and bottom distance 0.

Abasz, 2025-04-18:
I confirm that I experience the same issue

Ataraxia-Mechanica, 2025-07-18:
nuh uh uh

Abasz, 2025-10-18:
This is till an issue and has not been fixed.