OrcaSlicer/OrcaSlicer #8889: Text scaling in Device Tab is not persisted

bortek, 2025-03-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Voron 2 user her. After eash slicing and then switching to Device Tab view, text size there is very huge and I press CTRL+- 3 times to make it more readable and confortable. Now this has to be done every time after slicing or restarting OrcaSlicer and is not persisted. 

It is some ort of Microsoft Edge overlay that is used to display Device which is in my case is Mainsail

This probably affects many other printers too (but not Bumbu) since it is not using Edge overlay.

I have tried to open Mainsail in Edge and scale it down with CTRL+- and that is persisted because Edge remembers "per site setting" but not in the Device Tab. That is sort of re-initilized at every start and/or new slicing.

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

What I am looking for is either
1)  a way to persist whatever adjustment/scaling is done in Device tab
or
2) Make a default scaling 3 times less
or
3) Make default scaling configurable in some way

### Describe alternatives you've considered

See above

thelegendtubaguy, 2026-11-09:
I think a portion of this was fixed by #16752 and shipped in 2.5.0.   What's left is keeping zoom across restarts or a configurable default.

bortek, 2026-11-09:
Ia have installed 2.5 yesterday, which is BTW awesome quick to start.

Let me check and get back on this.

bortek, 2026-11-09:
Zoom adjustments in Decice tab are still not persisted across restarts of Orcaslicer.