OrcaSlicer/OrcaSlicer #8895: Crash when edit filament

real-artswan, 2025-03-14:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Linux

### OS Version

Ubuntu 24.04.2 LTS

### Printer

Voron Trident 350

### How to reproduce

1. Click to edit filament
2. Crash

### Actual results

The app is crashed and closed

### Expected results

It must open filament profile edit screen

### Project file & Debug log uploads

any project, independet

### Checklist of files to include

- [ ] Log file
- [ ] Project file

real-artswan, 2025-03-14:
The workaround is to change printer in the printer selection toolbox then change it again to the needed one.

real-artswan, 2025-03-14:
The issue is consistence from the very early versions, It starts to happen randomly but when it is started it's constantly.  I don't know how to debug AppImage app and where to look to debug info, if anybody can help I'll appreciate.

real-artswan, 2025-03-14:
I can unpack appimage but then what do you need? should i run it like gdb ./Orca_executable_whatever or what?
The issue is very annoying.

Noisyfox, 2025-03-15:
Could you upload the entire Orca configuration folder?

real-artswan, 2025-03-15:
@Noisyfox may I share it to you via GoogleDrive? It's 95Mb zipped

Noisyfox, 2025-03-15:
> [@Noisyfox](https://github.com/Noisyfox) may I share it to you via GoogleDrive? It's 95Mb zipped

Sure thing!

real-artswan, 2025-03-15:
Done for your email from the profile

Noisyfox, 2025-03-16:
If you go to `/var/crash` dir, is there anything there?

Noisyfox, 2025-03-16:
Also try `journalctl` command to see if there is any crash details.

real-artswan, 2025-03-16:
Thank you, will do when it happen.

real-artswan, 2025-03-19:
Just crashed right now again and nothing related in `jornalctl`, `/var/crash`, `/var/lib/apport/coredump`, etc. Also the standard "app crashed" window is not appeared. Orca just closes immediately when I press edit filament button

real-artswan, 2025-03-19:
Mounted AppImage and run under gdb. The following happens after button is pressed

`(orca-slicer:513541): Gtk-CRITICAL **: 19:55:28.416: gtk_window_resize: assertion 'height > 0' failed

Thread 1 "orcaslicer_main" received signal SIGSEGV, Segmentation fault.
_gtk_text_btree_find_line_top.isra.0 (target_line=0x555563850da0, view_id=<optimized out>, tree=<optimized out>) at ../../../gtk/gtktextbtree.c:1498
warning: 1498   ../../../gtk/gtktextbtree.c: No such file or directory
(gdb) 
`

[image]

Noisyfox, 2025-03-20:
Are you able to get the stacktrace by running the `bt` command in GDB when the crash occurs? Thanks!

real-artswan, 2025-03-21:
It's from the latest 2.3.0 release

```
Thread 1 "orcaslicer_main" received signal SIGABRT, Aborted.
Download failed: Invalid argument.  Continuing without source file ./nptl/./nptl/pthread_kill.c.
__pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
warning: 44     ./nptl/pthread_kill.c: No such file or directory
(gdb) bt
#0  __pthread_kill_implementation (no_tid=0, signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:44
#1  __pthread_kill_internal (signo=6, threadid=<optimized out>) at ./nptl/pthread_kill.c:78
#2  __GI___pthread_kill (threadid=<optimized out>, signo=signo@entry=6) at ./nptl/pthread_kill.c:89
#3  0x00007fffefa4527e in __GI_raise (sig=sig@entry=6) at ../sysdeps/posix/raise.c:26
#4  0x00007fffefa288ff in __GI_abort () at ./stdlib/abort.c:79
#5  0x00007ffff6cf4f5b in g_assertion_message
    (domain=domain@entry=0x7ffff7631014 "Gtk", file=file@entry=0x7ffff7648834 "../../../gtk/gtktextbtree.c", line=line@entry=1518, func=func@entry=0x7ffff76c46d0 <__func__.49> "_gtk_text_btree_find_line_top", message=message@entry=0x555563847300 "assertion failed: (iter->next != NULL)") at ../../../glib/gtestutils.c:3331
#6  0x00007ffff6d5da97 in g_assertion_message_expr
    (domain=domain@entry=0x7ffff7631014 "Gtk", file=file@entry=0x7ffff7648834 "../../../gtk/gtktextbtree.c", line=line@entry=1518, func=func@entry=0x7ffff76c46d0 <__func__.49> "_gtk_text_btree_find_line_top", expr=expr@entry=0x7ffff765280f "iter->next != NULL") at ../../../glib/gtestutils.c:3357
#7  0x00007ffff7624268 in _gtk_text_btree_find_line_top.isra.0 (target_line=0x555563874220, view_id=<optimized out>, tree=<optimized out>) at ../../../gtk/gtktextbtree.c:1518
#8  0x00007ffff74e07de in text_layout_changed (cursors_only=0, new_height=0, old_height=0, y=0, layout=0x55556693faf0) at ../../../gtk/gtktextbuffer.c:841
#9  gtk_text_layout_changed (layout=0x55556693faf0, y=0, old_height=0, new_height=0) at ../../../gtk/gtktextlayout.c:663
#10 0x00007ffff74c1902 in redisplay_region (tree=tree@entry=0x5555668f0830, start=<optimized out>, start@entry=0x7ffffffef030, end=<optimized out>, end@entry=0x7ffffffeefe0, cursors_only=cursors_only@entry=0)
    at ../../../gtk/gtktextbtree.c:2673
#11 0x00007ffff74c1a20 in queue_tag_redisplay (tag=<optimized out>, end=0x7ffffffeefe0, start=0x7ffffffef030, tree=0x5555668f0830) at ../../../gtk/gtktextbtree.c:1762
#12 0x00007ffff74c1c2e in _gtk_text_btree_tag (start_orig=0x555563854650, end_orig=0x5555664025d0, tag=0x5555663bf590, add=1) at ../../../gtk/gtktextbtree.c:1810
#17 0x00007ffff6e53883 in <emit signal 'apply-tag' on instance ???> (instance=instance@entry=0x555566407a90, signal_id=<optimized out>, detail=detail@entry=0) at ../../../gobject/gsignal.c:3583
    #13 0x00007ffff729aaa7 in _gtk_marshal_VOID__OBJECT_BOXED_BOXEDv
    (closure=<optimized out>, return_value=<optimized out>, instance=0x555566407a90, args=<optimized out>, marshal_data=<optimized out>, n_params=<optimized out>, param_types=0x5555619d7ea0)
    at gtk/gtkmarshalers.c:5244
    #14 0x00007ffff6e536bd in _g_closure_invoke_va (param_types=0x5555619d7ea0, n_params=<optimized out>, args=0x7ffffffef3d0, instance=0x555566407a90, return_value=0x0, closure=0x5555619d7ff0)
    at ../../../gobject/gclosure.c:897
    #15 signal_emit_valist_unlocked (instance=instance@entry=0x555566407a90, signal_id=signal_id@entry=387, detail=detail@entry=0, var_args=var_args@entry=0x7ffffffef3d0) at ../../../gobject/gsignal.c:3424
    #16 0x00007ffff6e537c1 in g_signal_emit_valist (instance=0x555566407a90, signal_id=387, detail=0, var_args=var_args@entry=0x7ffffffef3d0) at ../../../gobject/gsignal.c:3263
#18 0x00007ffff74c8c2b in gtk_text_buffer_emit_tag (buffer=0x555566407a90, tag=0x5555663bf590, apply=1, start=<optimized out>, end=<optimized out>) at ../../../gtk/gtktextbuffer.c:2611
#19 0x0000555559b96e6c in wxGtkTextApplyTagsFromAttr(_GtkWidget*, _GtkTextBuffer*, wxTextAttr const&, _GtkTextIter*, _GtkTextIter*) ()
#20 0x0000555559b97735 in au_insert_text_callback ()
#21 0x00007ffff6e332fa in g_closure_invoke (closure=0x555563847540, return_value=0x0, n_param_values=4, param_values=0x7ffffffefb50, invocation_hint=0x7ffffffefaa0) at ../../../gobject/gclosure.c:834
#22 0x00007ffff6e62c79 in signal_emit_unlocked_R.isra.0
    (node=node@entry=0x7ffffffefc70, detail=detail@entry=0, instance=instance@entry=0x555566407a90, emission_return=emission_return@entry=0x0, instance_and_params=instance_and_params@entry=0x7ffffffefb50)
    at ../../../gobject/gsignal.c:3961
#23 0x00007ffff6e53591 in signal_emit_valist_unlocked (instance=instance@entry=0x555566407a90, signal_id=signal_id@entry=379, detail=detail@entry=0, var_args=var_args@entry=0x7ffffffefde0)
    at ../../../gobject/gsignal.c:3520
#24 0x00007ffff6e537c1 in g_signal_emit_valist (instance=0x555566407a90, signal_id=379, detail=0, var_args=var_args@entry=0x7ffffffefde0) at ../../../gobject/gsignal.c:3263
#25 0x00007ffff6e53883 in g_signal_emit (instance=<optimized out>, signal_id=<optimized out>, detail=<optimized out>) at ../../../gobject/gsignal.c:3583
#26 0x0000555559b95fcd in wxTextCtrl::WriteText(wxString const&) ()
#27 0x0000555559b9973d in wxTextEntry::DoSetValue(wxString const&, int) ()
#28 0x0000555557067d35 in Slic3r::GUI::TextCtrl::set_value(boost::any const&, bool) ()
#29 0x000055555707d52e in Slic3r::GUI::ConfigOptionsGroup::reload_config() ()
#30 0x0000555556f88e1a in Slic3r::GUI::Page::reload_config() ()
[... cut ...]
#90 0x00007ffff6d32f77 in g_main_loop_run (loop=0x55555becc800) at ../../../glib/gmain.c:4419
#91 0x00007ffff73feb45 in gtk_main () at ../../../gtk/gtkmain.c:1329
#92 0x0000555559b05615 in wxGUIEventLoop::DoRun() ()
#93 0x0000555559c312b1 in wxEventLoopBase::Run() ()
#94 0x0000555559bfb8df in wxAppConsoleBase::OnRun() ()
#95 0x0000555559c82fc8 in wxEntry(int&, wchar_t**) ()
#96 0x0000555556d137eb in Slic3r::GUI::GUI_Run(Slic3r::GUI::GUI_InitParams&) ()
#97 0x00005555560d6c6b in Slic3r::CLI::run(int, char**) ()
#98 0x0000555555ff7958 in main ()
```

real-artswan, 2025-03-21:
It is not always so, sometimes it crashes immediately, restart the app doesn't help until you switch printer profile. I use AppImage

real-artswan, 2025-03-21:
But may be you're right, the problem happens if you close filamenet dialog not on the first tab. I'll keep eyes on that.

Noisyfox, 2025-03-22:
The problem now is the app crashed inside gtk, so I'm not sure what really happened or how to fix it.

Noisyfox, 2025-03-22:
I tried and everything seems valid. Something just doesn't work proerly inside gtk when setting the value of the `adaptive_pressure_advance_model` field. If you don't enable adaptive pa, it won't crash.

Noisyfox, 2025-03-22:
I think I've found a solution, will raise a PR soon.

Noisyfox, 2025-03-22:
Please give the PR #9016 a try and see if it fixes the problem.

real-artswan, 2025-03-23:
Wow, I'll try asap

real-artswan, 2025-03-23:
> when setting the value of the adaptive_pressure_advance_model field
afair it happened even before this feature was implemented, but will see.
Thanks for you work!

real-artswan, 2025-03-23:
@Noisyfox sorry, is it in current nightly build, right?

Noisyfox, 2025-03-23:
> [@Noisyfox](https://github.com/Noisyfox) sorry, is it in current nightly build, right?

Not yet. You could download the build from the pr.

real-artswan, 2025-03-24:
@Noisyfox I'm sorry, I'm not familiar with Orca's build system, how to dowload it? I can see from the github-action it was autocompiled and uploaded somewhere, but where it's a question to me. Please point.

real-artswan, 2025-03-24:
I mean where it is? name, label? How to access?
https://uploads.github.com/repos/SoftFever/OrcaSlicer/releases/137995723/assets{?name,label}

Noisyfox, 2025-03-24:
Go to that PR, there is a "Checks" tab on the top of the page. Click that, then there should be an "Artifacts" dropdown which you could find all the test builds.

real-artswan, 2025-03-29:
@Noisyfox You are a hero! 
I can't be sure because I doesn't have stable steps to reproduce, but I've been using the window and different tabs there pretty hard 2 days now without any fail. 
I beleive it's fixed.

real-artswan, 2025-03-29:
Should I do anything to get your PR into mainstream? It definitely helps and doesn't hurt. Then it can be reopened or so...

real-artswan, 2025-03-29:
I found another issue - when you open an older project where a filament profile isn't fresh. And there is an adaptive PA is set in the fresh one. You click refresh on the filament profile and it sets the checkbox but doesn't open the textbox with PA values. But it's the same on the your fix and on the current release. I'm not sure but may be this PR must be merged before the fix for the mentioned issue.

Noisyfox, 2025-03-30:
> I found another issue - when you open an older project where a filament profile isn't fresh. And there is an adaptive PA is set in the fresh one. You click refresh on the filament profile and it sets the checkbox but doesn't open the textbox with PA values. But it's the same on the your fix and on the current release. I'm not sure but may be this PR must be merged before the fix for the mentioned issue or all together. idk

I could reproduce this issue. Will take a look soon.

PR #9016 (2025-03-30): Fix filament editor crash on Linux
 Fix #8895

In some cases, updating multiline text fields during dialog pop up crashes GTK. This PR changes the order by updating the fields first then show the UI.
closed 2025-03-30 by SoftFever