OrcaSlicer/OrcaSlicer #8897: New Infill pattern

silversword411, 2025-03-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Saw this

https://scitechdaily.com/13x-stiffer-than-existing-designs-scientists-develop-new-material-with-remarkable-compressive-strength/%3E

I wonder if there are other infill patterns inspired by nature we can look into adding?

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

New infill pattern

L2on1, 2025-03-14:
Hi, unfortunately, the page seems to not exist anymore.

[image]

brocc28, 2025-03-14:
https://scitechdaily.com/13x-stiffer-than-existing-designs-scientists-develop-new-material-with-remarkable-compressive-strength/

Extra characters were added to the end of the URL. 
@silversword411

silversword411, 2025-03-14:
> characters were added to the end of the URL.

Thx! Copied a LT probably from Discord :D I updated origina post though

johnny-bit, 2025-03-15:
The paper is https://www.sciencedirect.com/science/article/pii/S0263822324009632