OrcaSlicer/OrcaSlicer #8900: Add option for "resin-style" supports optimized for FDM printing

envy-ai, 2025-03-14:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Just the usual problems with supports.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

A new support style that adds resin-style supports (as opposed to trees or standard supports).  Here's a video demonstrating how well it works:

https://www.youtube.com/watch?v=7kzJ0QSltkU

### Describe alternatives you've considered

This can be done manually, per the video, but it's a pain.

Mart7Mcfl7, 2025-03-14:
I'd love to see this, this with arc overhangs would make damn fine mini's

timnolte, 2025-03-24:
Even though I have been able to get organic tree supports dialed in pretty well I would love to see this option available. Something that would also be nice to see is a better auto-orientation to go along with it. Really if the auto-orient option would take into account the supports style/settings would be amazing.

UnrealKazu, 2025-04-14:
I've come to this ticket after having used resin supports manually through Lychee Slicer. It really is a game changer regarding the printing of mini's. In the past I've tried to mimic the behaviour of resin supports through trees, but never could fully get it to the point where it's as thin as a resin support. Now that I've printed with them for the first time, I see what a massive difference it makes with regards to ease of removal and object damage.

Having this feature in OrcaSlicer would definitely make it even better specifically for slicing min's.

mlewis-everley, 2025-09-19:
I would also love this feature! I have got some pretty good results with tree supports, but sometimes the tips and interface layers can destroy fine details.