OrcaSlicer/OrcaSlicer #8909: Wall printing order incorrect in some geometries (3 years old bug...)

rafpigna, 2025-03-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0 but also in all releases from at least 1.6.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

Creality K2 Plus, Creality K1 Max, Anycubic Kobra 3, Bambulab P1S, Bambulab X1C

### How to reproduce

Selecting wall printing order as "Internal-External" or "Internal-External-Internal", the sliced model doesnt follow this setting and walls are printed in External->Internal sequence.

This happens both with Arachne or Classic wall generators, no difference for me. 

This bug is present from at least version 1.6.0 and posted here -> #884 and also here -> #7145 but both were closed without any fix or comments.

Seems that this happens only with some concave/convex geometries, because if I drop a simple cube or cylinder on the plate, the wall printing order is correct, while the other model still shows the bug.

Attached there is the STL that I can't print in the right wall order.
[self-centering_roller_608_v2.zip](https://github.com/user-attachments/files/19257434/self-centering_roller_608_v2.zip)

### Actual results

Here is the video showing the printing simulation for the object generating the bug (the first one that is being printed in the video) and another object (cube) that is being printer after the first and shows correct behaviour in wall printing order. 

[image]

### Expected results

I expect that the sliced model follow the wall printing order I choose in the settings.

### Project file & Debug log uploads

[orca.zip](https://github.com/user-attachments/files/19257594/orca.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-15:
This is expected behavior, when the wall is too thin that cannot fit 2x wall count. The inner walls generated are actually generated as part of the hole, not the contour; and since we print contour first then holes, and the contour does not have any inner walls, so the outer wall is printed first.

A possible solution is when the wall is too thin, try generating inner walls as part of the contour as much as possible. This won't solve every possible cases (for example if the hole is not symmetrical, sometimes it will be hard to determine which wall belongs to which:
[image]

staal54a, 2025-05-22:
@Noisyfox I think you are right about the holes in the examples above and how they should print. However, I am seeing behavior on 2.3.0 where a closed geometry prints inner walls first when the feature is very thin. See screenshot below:

[image]

And here is the project file if you want to see for yourself: 

[wall_order_bug_m5.zip](https://github.com/user-attachments/files/20401967/wall_order_bug_m5.zip)

Evil-Crab, 2026-03-25:
Yeah, the issue is still present. For example:

[image]

[image]

[image]

Florin-Popescu, 2026-10-31:
Issue still present for me