OrcaSlicer/OrcaSlicer #8910: Pitch: configuration overlays

GarthSnyder, 2025-03-15:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

There is currently no way to memorialize incremental tweaks of a standard configuration for later reuse. This is a longstanding weakness of the Slic3r lineage that I don't think any fork has ever attempted to address. 

An illustrative example: you normally print with Voron-style settings, which are relatively robust. But for prototyping, you want just 2 walls, and the fill percentage should be as low as you can get away with (or use a different type of fill entirely, e.g., Lightning). Also, prototypes don't need regular quality, so speeds and accelerations can be raised in the service of rapid iteration.

Currently (and please correct me if I'm wrong about this), there's no way to capture such a "prototyping" configuration other than to modify your existing profiles and save them under different names. You might need to save multiple profiles this way. For example, if you're creating a "printing with 0.25mm nozzle" configuration, you'll need new, separate profiles for both the print settings and the printer.

Once you do that, though, there's no ongoing connection between profiles. If you want to change something in your standard configuration, you have to hunt down every variant profile and edit it by hand. Or just let the profiles gradually drift apart like tectonic plates, and make do.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

This issue has been discussed in multiple threads on multiple projects. For example, here's [a nice suggestion](https://github.com/prusa3d/PrusaSlicer/issues/7983) filed against PrusaSlicer by @OfficialDyray and [a roll-up of past discussions](https://github.com/supermerill/SuperSlicer/issues/4232) posted by @Andrei-Pozolotin against SuperSlicer. Well worth reading!

There seems to be a general consensus that "Yes, someone should address that problem!" But it never actually progresses to development because a) it's likely quite difficult, and b) it's not something that has an obvious UI. There are a variety of possible approaches, but the potential for crapifying the UI is always there.

Is anyone working on this kind of extension for OrcaSlicer? I'm tempted to give it a go myself, but I don't want to step on anyone's toes.

Noisyfox, 2025-03-15:
Does profile inherit solve this problem? If we allow user profile to inherit from another user profile, so child profile only stores parameters that differ from its parent. Then in this way if you update anything to the parent profile, as long as the child profile does not override that same option, then this new value should be applied to all children/grand-children/etc. automatically.

GarthSnyder, 2025-03-15:
> Does profile inherit solve this problem?

It well might! I hadn't realized that this was the default behavior, but I can see that OrcaSlicer is generating new profiles with inheritance. Let me investigate further and I'll report back here.

Noisyfox, 2025-03-16:
Currently it's not possible to create grandchildren profiles unfortunately, even if you manually edited the json files, due to an issue in current profile loading process. We've been talking about improving this privately for a while now, hut haven't come up with a proper solution yet.

GarthSnyder, 2025-03-18:
I'm still not sure I have the whole lineage figured out, but it looks like PrusaSlicer has never supported user-level inheritance, although it is allowed within system configuration bundles.

OrcaSlicer currently seems to allow a single level of inheritance at user level. If a modified profile that is being saved under a different name already inherits from a third profile, OrcaSlicer saves the new profile based on that same parent profile, storing only the deltas between the parent and both user-level profiles. (That is, if Profile A inherits from System Profile X, and Profile A is then modified and saved as a new Profile B, Profile B's JSON file names System Profile X as the parent and includes the modifications from both Profile A and Profile B. Or said another way, at the end of the update, Profile B looks exactly the way Profile A would look if it had been saved under its own name.) I'm not sure if this behavior is new with OrcaSlicer or whether it's inherited from Bambu Studio.

OrcaSlicer seems to handle user-level profile inheritance correctly, too. For example, if Profile A in the example above does not inherit from another profile, Profile B then inherits directly from Profile A, and only the differences between A and B are saved in B's JSON file. The UI seems to handle this setup (one user-level profile inheriting from another user-level profile) pretty smoothly. Changes to the base profile, once saved, are immediately reflected in the UI's presentation of the child profile.

As you mention, the UI won't create profile chains. It only deals with a single level of inheritance. And in fact, if you manually edit configurations to introduce a chain, the UI seems to get quite confused.

This all seems quite promising and 90% of the way there already.  I think the ability to insert one more level of inheritance would be helpful, since it would allow profiles to inherit both from the system-supplied configurations and from a user-maintained profile.

That doesn't necessarily mean inheritance needs to be fully general and nestable, though (although that would be nice). If two levels are easier to handle in some ad-hoc way that doesn't fully generalize, I still think it may be worth doing.

It would also be helpful if you could specify the profile you want to inherit from when saving. Profiles in the existing chain of inheritance would be obvious choices, but it'd also be nice if you could specify an arbitrary third profile. That would allow users to restructure their profile tree without adding much UI. It would also let you port collections of older flat profiles into a hierarchy.

Noisyfox, 2025-03-18:
> OrcaSlicer seems to handle user-level profile inheritance correctly, too.

Not always, due to a bug during the user profile loading mechanism. Say you have 2 user profiles A and B, and B.inherit is set to A. Then it will only works properly if A has been loaded before B. However the loading order is undetermined (mostly controlled by which one is returned first by the file system when iterating through the user profile folder), so if B is loaded first, then when it try to apply deltas to its parent profile, A, it will not be able to do that, it will instead fallback to use a default system profile. This is exactly what happend in issue #8366.

Noisyfox, 2025-03-18:
> It would also be helpful if you could specify the profile you want to inherit from when saving. Profiles in the existing chain of inheritance would be obvious choices, but it'd also be nice if you [...]

Might be able to do this since it doesn't change any of the existing profile loading process. Though the loading issue must be solved first before allowing creating nested inherit.

GarthSnyder, 2025-03-19:
Let me know if there's anything I can do to help!