OrcaSlicer/OrcaSlicer #8918: Poor layer adhesion when printing PETG layers with high speed due to material change delays

shurf-dnepr, 2025-03-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU AMD Ryzen 7 3700
RAM 32 Gb
Video Card Nvidia GeForce RTX 2070 Super

### Printer

Bambulab A1 Combo (AMS Lite)

### How to reproduce

1. Open OrcaSlicer (v2.2.0).
2. Configure a dual-material print:

- Use PLA for support interface layers.
- Use PETG for the main model.

3. Enable automatic material switching (e.g., on Bambu Lab A1 with AMS).
4. Slice a model where material changes occur during the print.
5. Observe the behavior of PETG layers printed at the same height as the PLA interface layer.

### Actual results

**Actual Behavior :**

During material changes, the slicer calculates the print speed based on the total layer time, including the time taken for material switching.
For example:
A normal layer takes 10 seconds to print, resulting in a reasonable volumetric speed (e.g., 5 mm³/s).
A layer with a material change takes 320 seconds due to the filament swap delay.
The slicer compensates by increasing the print speed to the maximum allowed volumetric speed (e.g., 40 mm³/s), which exceeds the PETG material's recommended limit (9 mm³/s).
As a result, the extruder fails to fully melt the PETG filament, leading to under-extrusion and poor inter-layer adhesion.

**Workaround :**
Disabling part cooling fans for affected layers partially mitigates the issue, but it does not fully resolve the problem.

[image]
[image]
[image]
[image]
[image]

### Expected results

**Expected Behavior :**

PETG layers should maintain proper inter-layer adhesion regardless of material changes.
The slicer should respect the maximum volumetric speed settings for PETG (e.g., 9 mm³/s) even during material changes.

### Project file & Debug log uploads

[Poor layer adhesion when printing PETG layers with high speed due to material change delays.zip](https://github.com/user-attachments/files/19264556/Poor.layer.adhesion.when.printing.PETG.layers.with.high.speed.due.to.material.change.delays.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-16:
> During material changes, the slicer calculates the print speed based on the total layer time, including the time taken for material switching.

This issue should have been fixed in 2.3, please give it a try.

shurf-dnepr, 2025-03-16:
> > During material changes, the slicer calculates the print speed based on the total layer time, including the time taken for material switching.
> 
> This issue should have been fixed in 2.3, please give it a try.

Will test it immediately, thanks!

shurf-dnepr, 2025-03-16:
> please give it a try.

Unfortunately the situation has not changed. Volume flow and speed still take into account the total printing time of the layer, together with the time for plastic replacement. ORCA Slicer version 2.3.0-rc nightly build

[image]
[image]

Noisyfox, 2025-03-16:
Ah you're right. My previous fix works only if the toolchange happens at the very begining of the layer.

Noisyfox, 2025-03-16:
[image] For now you could disable this option as a workaround. Though it will make the entire print faster:
[image]

Noisyfox, 2025-03-16:
I would suggest that, if higher speed significantly reduces the layer adhesion, then it's better to turn the speed down globally, instead of relying on the auto-slowdown.

shurf-dnepr, 2025-03-16:
> For now you could disable this option as a workaround. Though it will make the entire print faster:

This feature is always enabled in my settings. If it is turned off, the cooling will not always keep up with this printing speed and the part may warp.

shurf-dnepr, 2025-03-16:
> Я б запропонував, що якщо більша швидкість значно зменшує адгезію шару, то краще зменшити швидкість глобально, 

Indeed, you're right. Global reduction of printing speed solves this problem, but unfortunately it creates another one - printing time of the product increases significantly ).

Noisyfox, 2025-03-16:
> The slicer compensates by increasing the print speed to the maximum allowed volumetric speed (e.g., 40 mm³/s), which exceeds the PETG material's recommended limit (9 mm³/s).

Though I don't think this really happens according to the gcode.

shurf-dnepr, 2025-03-16:
> > The slicer compensates by increasing the print speed to the maximum allowed volumetric speed (e.g., 40 mm³/s), which exceeds the PETG material's recommended limit (9 mm³/s).
> 
> Though I don't think this really happens according to the gcode.

I agree with that, couldn't reproduce it in my place. If necessary, I can try to remove it from the problem description.

shurf-dnepr, 2025-03-16:
Perhaps there is a sense to add a separate parameter, setting the printing speed of layers on which there is an interface layer? As an option, if the user has chosen in the settings to print the interface layer with an alternative material, then on the layers with the interface support, just forced to use the average print speed. Average layer print speed = 1 layer preceding the interface layer + 1 layer following the last interface layer.

Noisyfox, 2025-03-16:
Or maybe just simply exclude filament change time from the equation entirely.

shurf-dnepr, 2025-03-16:
> Or maybe just simply exclude filament change time from the equation entirely.

That would be awesome! Perhaps we should also exclude the printing time of the interface layer from the calculation, as the printing time there can be radically different from the printing time of the main part

shurf-dnepr, 2025-04-15:
> Or maybe just simply exclude filament change time from the equation entirely.

Good afternoon! Could you please tell me when we can expect this issue to be fixed?

shurf-dnepr, 2025-04-18:
> Though I do think slowing down print speed for cooling is generally bad for quality

Yes, I completely agree that forced speed reduction can have a bad effect on the product. 
In our case, we won't force changes in print speed, we will only remove from the print speed calculations the time that is not relevant to printing and taking into account the context of printing, i.e. previous and subsequent layers, the resulting print speed result will perfectly match them. 
As soon as this improvement appears, I'm sure that almost all bambulab studio users will finally switch to ORCA.
As well as users of other printers with MMU systems)