OrcaSlicer/OrcaSlicer #8920: Auto-Arrange on Toolbar

Tombeam, 2025-03-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-RC

### Operating System (OS)

Windows

### OS Version

Windows 11

### Additional system information

Acer Aspire 5 - R7 5700U - 16GB (Aspire 3 as well with 40GB ram)

### Printer

Creality K2 Plus - 1.1.2.6

### How to reproduce

Clicking the auto-arrange button on the toolbar results in the plate lock/unlock icon to flash, and returns an error, cannot auto-arrange locked plate. If you click the arrange icon beside each plate, it works just fine.

Using 3 colors and print by object.

[image]

### Actual results

see above?

### Expected results

see above?

### Project file & Debug log uploads

[log-2-15-2025.zip](https://github.com/user-attachments/files/19265379/log-2-15-2025.zip)

[brake gauge.zip](https://github.com/user-attachments/files/19265408/brake.gauge.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Tombeam, 2025-03-15:
only seems to be when you select print by object. selected from the left process>objects tab

Noisyfox, 2025-03-16:
Can reproduce. Will take a look soon.

Noisyfox, 2025-03-16:
It seems only happen when you set the print sequence on the plate instead of globally. Will see if this can be fixed.

Tombeam, 2025-08-26:
I agree with the better error code, it doesn't really seem to be a bug once explained.