OrcaSlicer/OrcaSlicer #8921: Multidisplay corruption / GUI bug when resizing the split view (config pane) border

malventano, 2025-03-15:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Windows

### OS Version

11

### Printer

N/A

### How to reproduce

Attempt to resize the config pane by dragging the divider / border left or right.

### Actual results

- Note the UI does not update while dragging and only snaps to the new position when releasing.
- Note that if using multiple displays, and Orca is on the right display, graphical artifacts are drawn on the left display as the border is dragged. The artifacts perisist until the display is refreshed. Example below:

[image]

### Expected results

UI should update while dragging the divider.
Artifacts should not be drawn outside of the OrcaSlicer window, and especially not drawn onto other displays.

### Project file & Debug log uploads

N/A

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

When using single display, the UI still does not update while dragging the divider.

tlhintoq, 2025-03-17:
I have what sounds like a very similar monitor arrangement.
I also have Orca on the right
There's no artifacts or noise while dragging.

[image]

> UI should update while dragging the divider.
I doubt that's ever going to happen.  Its not in the nature of the cross-platform framework used in this program.  And there's plenty of other programs that have the exact same behavior.  Its not a bug, its just how that framework is.

> Artifacts should not be drawn outside of the OrcaSlicer window, and especially not drawn onto other displays.
Sounds like you need to set Orca to be using the high-performance video card by default.  Windows it probably launching it on the default chipset.

[image]

malventano, 2025-03-17:
> I have what sounds like a very similar monitor arrangement.

They appear to be quite different.

> I also have Orca on the right There's no artifacts or noise while dragging.

The displays are not on the same horizontal plane and are not the same resolution. This makes symptom elaboration incompatible across both systems as in my configuration the accelerated draw calls are being directed to the wrong display.

> Its not in the nature of the cross-platform framework used in this program. And there's plenty of other programs that have the exact same behavior. Its not a bug, its just how that framework is.

QT / Electron support updates to the UI while dragging an element. The artifact appears to be the framework attempting to update the divider location in real-time, as the artifact looks to be an XOR of the UI contents, which would be drawn in the wake of the moving divider. This bug has existed since Orca added support for resizing the configuration pane, and is unique to Orca.

Additionally, Orca redraws the window contents while resizing the window itself, so clearly this is a capability of the framework and it is simply bugged when resizing the config pane.

> Sounds like you need to set Orca to be using the high-performance video card by default. Windows it probably launching it on the default chipset.

This is a Xeon 3495X Workstation system. There is no 'default chipset' graphics. Only the RTX 3090.
[image]

malventano, 2025-03-29:
I dropped down to single display and was able to see the intended behavior (a grey bar shows where the border will be placed):

[image]

The grey bar updates are what is being drawn to the wrong display in 2-display mode, with the primary display to the right.

tlhintoq, 2025-03-30:
> Resizing the config pane is a feature unique to Orca Slicer.
Haha

Armorsmith Designer
[image]

Lightburn
[image]

Darktable {I could go on and on}
[image]

closed 2025-07-07 by github-actions[bot]