OrcaSlicer/OrcaSlicer #8929: New flow rate calibration paterns

Benik3, 2025-03-16:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

With actual flow rate calibration models it can be hard to determine right flow, mainly with some specific filaments (e.g. crystal/transparent one)

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

I found by luck this flow rate calibration method: https://makerworld.com/en/models/189543-improved-flow-ratio-calibration-v3
In comparison to the actual calibration method, it's very easy to find correct flow even with filaments with hard "visual quality check".
The principle is simple, use archimedean chords patern on the top layer and let the model slice, that it prints first one side of the half circles and then it prints the middle helix from the middle (that's important). So the last circle of the helix must fit into the one line space.
Then it prints part of the second side and the rest is printed from oposite direction, which again ends with one line fit.
Sometimes I have to tinker with it a little (e.g. rotate the object), because sometimes the slicer start the helix from outside, not inside.
These pictures describe it better:

[image]

[image]

[image]

### Additional context

I successfully used this pattern also to calibrate first layer, I just had to manually edit the g-code to print objects with different z-offset. This request is already here in #7874

barndawe, 2025-03-16:
Seconded, I've been using this for a little while and found it much easier to calibrate with

Benik3, 2025-03-24:
Thanks! Is possible to download build of this PR or I have to build myself?

Benik3, 2025-04-07:
Yes, it's correct behavior because of different slicing to ensure that the helix is printed from center: https://github.com/SoftFever/OrcaSlicer/pull/9145

Yes, Jim's says that he uses the value, where the circle starts to "overflow". I used successfully the one, where it's flat.
Or the middle way, select value between these two :)
Higher flow can prevent gaps and can have better strength, but with the lower one you will get more details. So as always, it also depends on your preferences and usage.

This is how the test should look like:
[image]

As you can see, with too much flow, the last circle of the helix will start to overflow (build up), because it's not able to fit in the remaining gap (which should be exactly one line width).

About the line width, I also noticed that it's printed with 120%. In my opinion I would rather leave it to the standard width, so you calibrate with settings, which is really used for printing... But I tested both and got similar results (120% vs 105%).

llitz, 2025-04-07:
Ok I will give it another try with the latest code, mine isn't overflowing at all, but it wasn't printed on the latest release either.

uniqueacid, 2025-08-28:
OrcaSlicer V2.3.1-alpha printed calibration file does not have tile numbering, or am I doing something wrong?

[image]