OrcaSlicer/OrcaSlicer #8933: Support interface not  honoring XY distance resulting in interface with no support under it

tlhintoq, 2025-03-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-RC

### Operating System (OS)

Windows

### OS Version

10x64

### Additional system information

High end machine per all my other earlier posts

### Printer

Elegoo giga

### How to reproduce

With a model that might have support needs closer than the XY offset
Turn on supports.

### Actual results

[image]

The supports stop because they honor the 1mm distance from the model.
But the interface still tries to print even though there is no support under it.

### Expected results

If there's no support, don't try to print interface in mid air

### Project file & Debug log uploads

x

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

x

hotellonely, 2025-03-17:
oh, so it was indeed a bug not just some bad luck i got

tlhintoq, 2025-03-19:
My example was from a large job done on an Elegoo Giga with a 1.0mm nozzle.
I think this, like a lot of the support issues, become more obvious when the jobs are larger.

[image]

[Thighs.giga.orca (2).zip](https://github.com/user-attachments/files/19330443/Thighs.giga.orca.2.zip)

tlhintoq, 2025-03-19:
> oh, so it was indeed a bug not just some bad luck i got

Are you saying this happened to you as well?  I've suppled @Noisyfox with a `.3mf` project that produces this.  Maybe you could share yours as well?  That would give her a couple different projects to maybe find a common cause.

Noisyfox, 2025-03-21:
I've located the issue, however I couldn't find an easy solution. The issue is the support area is too small that, when generating extrusion paths, in one direction it fits but when rotated 90 degrees it won't. The support body and interface have a 90 degree angle difference, that's why only the interface is generated but not the body.

tlhintoq, 2025-03-25:
@Noisyfox 
> The support body and interface have a 90 degree angle difference,

So if the user were to ensure the support and interface settings are at the same angle this shouldn't happen?
Is that something the user even has the ability to control?

Noisyfox, 2025-03-25:
No you couldn't change them individually. And this is just one factor of the problem.

tlhintoq, 2025-03-31:
@Noisyfox 

Ummm... Maybe I'm on to something here?

I run a 1.0 nozzle on my giga.  That means I have a layer height range maxed at 0.8 - Which I only use for some draft quality work.

I've just realized that range is being used to combine support layer thickness, producing some pretty thick layers in places.
I'm wondering if this would account for the big holes/jumps in the tree supports (several tickets) as well as supports in mid air and so on.  

And I can't find any way to turn of support layer combining - so for now I'm testing with a max layer height of only .4 - I might even set the max to my current job height depending on how that goes.

[image]

Noisyfox, 2025-03-31:
> [@Noisyfox](https://github.com/Noisyfox)
>
> Ummm... Maybe I'm on to something here?
>
> I run a 1.0 nozzle on my giga. That means I have a layer height range maxed at 0.8 - Which I only use for some [...]

Could you provide the project file?

closed 2025-07-07 by github-actions[bot]