OrcaSlicer/OrcaSlicer #8935: Unable to select custom filament in AMS

SoCuul, 2025-03-17:
I have many custom filaments that were created through the "Custom Filament" GUI, but when I try to select them in the AMS I get the following popup error

[image]\

Looking into the code that produces this error, I found these checks. However, my filament profile should provide all the values required for this to not happen.

https://github.com/SoftFever/OrcaSlicer/blob/4eaef65a7904b2763ff9b7a6dfaccec790f8080f/src/slic3r/GUI/AMSMaterialsSetting.cpp#L568

Filament profile:
```json
{
    "activate_air_filtration": [
        "0"
    ],
    "activate_chamber_temp_control": [
        "0"
    ],
    "adaptive_pressure_advance": [
        "0"
    ],
    "adaptive_pressure_advance_bridges": [
        "0"
    ],
    "adaptive_pressure_advance_model": [
        "0,0,0\n0,0,0"
    ],
    "adaptive_pressure_advance_overhangs": [
        "0"
    ],
    "additional_cooling_fan_speed": [
        "0"
    ],
    "chamber_temperature": [
        "0"
    ],
    "close_fan_the_first_x_layers": [
        "1"
    ],
    "compatible_printers": [
        "Bambu Lab P1P 0.4 nozzle",
        "Bambu Lab P1S 0.4 nozzle"
    ],
    "compatible_printers_condition": "",
    "compatible_prints": [ ],
    "compatible_prints_condition": "",
    "complete_print_exhaust_fan_speed": [
        "70"
    ],
    "cool_plate_temp": [
        "35"
    ],
    "cool_plate_temp_initial_layer": [
        "35"
    ],
    "default_filament_colour": [
        "#161616"
    ],
    "dont_slow_down_outer_wall": [
        "0"
    ],
    "during_print_exhaust_fan_speed": [
        "70"
    ],
    "enable_overhang_bridge_fan": [
        "1"
    ],
    "enable_pressure_advance": [
        "1"
    ],
    "eng_plate_temp": [
        "0"
    ],
    "eng_plate_temp_initial_layer": [
        "0"
    ],
    "fan_cooling_layer_time": [
        "80"
    ],
    "fan_max_speed": [
        "100"
    ],
    "fan_min_speed": [
        "50"
    ],
    "filament_cooling_final_speed": [
        "3.4"
    ],
    "filament_cooling_initial_speed": [
        "2.2"
    ],
    "filament_cooling_moves": [
        "4"
    ],
    "filament_cost": [
        "22.4"
    ],
    "filament_density": [
        "1.24"
    ],
    "filament_deretraction_speed": [
        "nil"
    ],
    "filament_diameter": [
        "1.75"
    ],
    "filament_end_gcode": [
        "; filament end gcode \nM106 P3 S0\n"
    ],
    "filament_flow_ratio": [
        "0.981"
    ],
    "filament_id": "Pe43ecc9",
    "filament_is_support": [
        "0"
    ],
    "filament_loading_speed": [
        "28"
    ],
    "filament_loading_speed_start": [
        "3"
    ],
    "filament_long_retractions_when_cut": [
        "nil"
    ],
    "filament_max_volumetric_speed": [
        "12"
    ],
    "filament_minimal_purge_on_wipe_tower": [
        "15"
    ],
    "filament_multitool_ramming": [
        "0"
    ],
    "filament_multitool_ramming_flow": [
        "10"
    ],
    "filament_multitool_ramming_volume": [
        "10"
    ],
    "filament_notes": [
        ""
    ],
    "filament_ramming_parameters": [
        "120 100 6.6 6.8 7.2 7.6 7.9 8.2 8.7 9.4 9.9 10.0| 0.05 6.6 0.45 6.8 0.95 7.8 1.45 8.3 1.95 9.7 2.45 10 2.95 7.6 3.45 7.6 3.95 7.6 4.45 7.6 4.95 7.6"
    ],
    "filament_retract_before_wipe": [
        "nil"
    ],
    "filament_retract_lift_above": [
        "nil"
    ],
    "filament_retract_lift_below": [
        "nil"
    ],
    "filament_retract_lift_enforce": [
        "nil"
    ],
    "filament_retract_restart_extra": [
        "nil"
    ],
    "filament_retract_when_changing_layer": [
        "0"
    ],
    "filament_retraction_distances_when_cut": [
        "nil"
    ],
    "filament_retraction_length": [
        "nil"
    ],
    "filament_retraction_minimum_travel": [
        "nil"
    ],
    "filament_retraction_speed": [
        "nil"
    ],
    "filament_settings_id": [
        "SUNLU PLA Matte Black @ 0.4 nozzle"
    ],
    "filament_shrink": [
        "100%"
    ],
    "filament_shrinkage_compensation_z": [
        "100%"
    ],
    "filament_soluble": [
        "0"
    ],
    "filament_stamping_distance": [
        "0"
    ],
    "filament_stamping_loading_speed": [
        "0"
    ],
    "filament_start_gcode": [
        "; filament start gcode\n{if  (bed_temperature[current_extruder] >45)||(bed_temperature_initial_layer[current_extruder] >45)}M106 P3 S255\n{elsif(bed_temperature[current_extruder] >35)||(bed_temperature_initial_layer[current_extruder] >35)}M106 P3 S180\n{endif};Prevent PLA from jamming\n\n\n{if activate_air_filtration[current_extruder] && support_air_filtration}\nM106 P3 S{during_print_exhaust_fan_speed_num[current_extruder]} \n{endif}"
    ],
    "filament_toolchange_delay": [
        "0"
    ],
    "filament_type": [
        "PLA"
    ],
    "filament_unloading_speed": [
        "90"
    ],
    "filament_unloading_speed_start": [
        "100"
    ],
    "filament_vendor": [
        "SUNLU"
    ],
    "filament_wipe": [
        "0"
    ],
    "filament_wipe_distance": [
        "nil"
    ],
    "filament_z_hop": [
        "nil"
    ],
    "filament_z_hop_types": [
        "nil"
    ],
    "from": "User",
    "full_fan_speed_layer": [
        "0"
    ],
    "hot_plate_temp": [
        "65"
    ],
    "hot_plate_temp_initial_layer": [
        "65"
    ],
    "idle_temperature": [
        "0"
    ],
    "inherits": "",
    "is_custom_defined": "0",
    "name": "SUNLU PLA Matte Black @ 0.4 nozzle",
    "nozzle_temperature": [
        "215"
    ],
    "nozzle_temperature_initial_layer": [
        "215"
    ],
    "nozzle_temperature_range_high": [
        "240"
    ],
    "nozzle_temperature_range_low": [
        "190"
    ],
    "overhang_fan_speed": [
        "100"
    ],
    "overhang_fan_threshold": [
        "50%"
    ],
    "pellet_flow_coefficient": [
        "0.4157"
    ],
    "pressure_advance": [
        "0.01"
    ],
    "reduce_fan_stop_start_freq": [
        "1"
    ],
    "required_nozzle_HRC": [
        "3"
    ],
    "slow_down_for_layer_cooling": [
        "1"
    ],
    "slow_down_layer_time": [
        "8"
    ],
    "slow_down_min_speed": [
        "20"
    ],
    "support_material_interface_fan_speed": [
        "-1"
    ],
    "temperature_vitrification": [
        "45"
    ],
    "textured_cool_plate_temp": [
        "40"
    ],
    "textured_cool_plate_temp_initial_layer": [
        "40"
    ],
    "textured_plate_temp": [
        "65"
    ],
    "textured_plate_temp_initial_layer": [
        "65"
    ],
    "version": "1.9.0.14"
}
```

SoftFever, 2025-06-17:
Hi
Have you tried the nightly build? It should have been fixed.
If you still see the issue, can you provide a step by step guide to reproduce the issue?

SoCuul, 2025-09-16:
> Hi Have you tried the nightly build? It should have been fixed. If you still see the issue, can you provide a step by step guide to reproduce the issue?

Apologies for the late response to this, but I was able to confirm that it is indeed fixed. Thanks!