OrcaSlicer/OrcaSlicer #8940: dep_OCCT PATCH_COMMAND failed while building OrcaSlicer V2.2.0 on Linux Debian 12 bookworm (on RPi4 aarch64)

dannutu, 2025-03-17:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

V2.2.0

### Operating System (OS)

Linux

### OS Version

Debian GNU/Linux 12 (bookworm)

### Additional system information

Linux raspberrypi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux
CPU: aarch64
Memory: 8GB RAM + 24GB swap (USB3 connected SSD)

### Printer

P1S

### How to reproduce

Download & extract .tar.gz from https://github.com/SoftFever/OrcaSlicer/archive/refs/tags/v2.2.0.tar.gz
```
sudo ./BuildLinux.sh -u
./BuildLinux.sh -dsir
```

### Actual results

```
[173/185] Performing patch step for 'dep_OCCT'
FAILED: dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch /home/bubu/Download/Orca-build-17Mar2025/OrcaSlicer-2.2.0/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
cd /home/bubu/Download/Orca-build-17Mar2025/OrcaSlicer-2.2.0/deps/build/dep_OCCT-prefix/src/dep_OCCT && git apply --directory deps/build/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix /home/bubu/Download/Orca-build-17Mar2025/OrcaSlicer-2.2.0/deps/OCCT/0001-OCCT-fix.patch && /usr/bin/cmake -E touch /home/bubu/Download/Orca-build-17Mar2025/OrcaSlicer-2.2.0/deps/build/dep_OCCT-prefix/src/dep_OCCT-stamp/dep_OCCT-patch
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/CMakeLists.txt: No such file or directory
Checking patch deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx...
error: deps/build/dep_OCCT-prefix/src/dep_OCCT/src/Font/Font_FTFont.cxx: No such file or directory
ninja: build stopped: subcommand failed.
```

### Expected results

successful build process

### Project file & Debug log uploads

OrcaSlicer build failure

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

I didn't have enough time to get to the root cause, however I managed to work around it by changing the line:
`PATCH_COMMAND git apply --directory ${BINARY_DIR_REL}/dep_OCCT-prefix/src/dep_OCCT --verbose --ignore-space-change --whitespace=fix ${CMAKE_CURRENT_LIST_DIR}/0001-OCCT-fix.patch`
to:
`PATCH_COMMAND git apply --verbose --ignore-space-change --whitespace=fix` ${CMAKE_CURRENT_LIST_DIR}/0001-OCCT-fix.patch
in the file deps/OCCT/OCCT.cmake
Then applying the 3 steps from here: https://github.com/SoftFever/OrcaSlicer/discussions/1425#discussioncomment-6451501

> You can rebuild OCCT only by:
> delete "deps\build\dep_OCCT-prefix" folder
> CD to "deps\build" folder, run cmake --build . --config Release --target dep_OCCT
> delete OrcaSlicer's build folder and build again.

I hope this could be addressed in the main branch for the upcoming V2.3.0.

Noisyfox, 2025-03-19:
Did you download the Orca's source code by git clone, or the zip archive?

Noisyfox, 2025-03-19:
I think it's the same as https://github.com/SoftFever/OrcaSlicer/pull/8465

closed 2025-03-19 by Noisyfox