OrcaSlicer/OrcaSlicer #8957: Repeated Crash on Flatpak v2.3.0 release candidate

Syyrion, 2025-03-18:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc (Flatpak)

### Operating System (OS)

Linux

### OS Version

Fedora 41

### Printer

Generic Klipper printer with Mainsail

### How to reproduce

1. Go to edit physical printer.
2. Enter a Hostname, IP, or URL
3. Press OK

### Actual results

Orca will crash immediately due to a segfault. Restarting Orca doesn't fix the issue as it will still crash on startup. You have to manually edit the config to remove the hostname from the printer before it will start normally again.

### Expected results

The device page should show the Mainsail web app.

### Project file & Debug log uploads

Nothing super interesting was found in the logs

[debug_Tue_Mar_18_12_01_41_2.log](https://github.com/user-attachments/files/19323184/debug_Tue_Mar_18_12_01_41_2.log)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

This is probably because of an issue when loading the mainsail web app on the device page.

frk2, 2025-03-18:
Experiencing the same using AppImage on Ubuntu 22.04. Started happening after configuring the hostname

Noisyfox, 2025-03-19:
Dup https://github.com/SoftFever/OrcaSlicer/issues/7210

closed 2025-03-19 by Noisyfox