OrcaSlicer/OrcaSlicer #8977: Octagram spiral solid infill does weird things

eternal-sorrow, 2025-03-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0-rc

### Operating System (OS)

Linux

### OS Version

Flatpak

### Additional system information

not applicable

### Printer

Elegoo Neptune 4 Max

### How to reproduce

1. Open any STL with wide horizontal surfaces
2. Set Bottom surface pattern/Top surface pattern/Solid infill pattern to Octagram Spiral
3. Slice plate
4. Drag the sliders in the preview tab to see how is the infill gonna be printed

### Actual results

Instead of the even continuous printing we get some weird patterns where it prints some strips, leaving the gaps between them, then returns to fill the gaps. This leads to pretty rough surfaces.

### Expected results

Continuously print the outer parts first, then fill in the inner spiral.

### Project file & Debug log uploads

[tray.zip](https://github.com/user-attachments/files/19361664/tray.zip)

[debug_Thu_Mar_20_16_37_51_2.log.0.zip](https://github.com/user-attachments/files/19361666/debug_Thu_Mar_20_16_37_51_2.log.0.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

[image]

[image]

[image]

[image]

[image]
Why does it do that?

Felix14-v2, 2025-03-20:
Works fine in OrcaSlicer 2.2.0

[image]

Ataraxia-Mechanica, 2025-03-22:
It's because this infill isn't (monotonic)[https://help.prusa3d.com/article/infill-patterns_177130] which should be default for all types of infill imo.

eternal-sorrow, 2025-03-22:
One thing is being monotonic, when some features break the infill pattern and the algorithm has to decide what parts of the broken pattern print first and what to print later. And the other thing when on the completely flat featureless surface it still does some guesswork and behaves erratically when it could be completely predictable.

iamnotahippy, 2025-04-11:
Rectilinear and Aligned Rectilinear infill patterns exhibit this behavior as well. This is a new issue, as I've used these patterns extensively in the past.

eternal-sorrow, 2025-07-11:
Not stale, the problem is still present. Automatic issue closing shouldn't be a thing.