OrcaSlicer/OrcaSlicer #8980: When going to the device tab, it closes after a few seconds

DiomedesDominguez, 2025-03-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Linux

### OS Version

Pop!_OS 22.04 LTS

### Additional system information

CPU: Intel® Core™ i7-9750H CPU @ 2.60GHz × 12
RAM: 64.0 GiB
Graphics: NVIDIA Corporation TU106M [GeForce RTX 2060 Mobile]

### Printer

Bambulab P1S

### How to reproduce

1. Go to the Device tab
2. Wait a few seconds

https://github.com/user-attachments/assets/85c459c6-3236-4b6c-83a7-4bee6564470e

### Actual results

It closes automatically

### Expected results

Not to close

### Project file & Debug log uploads

There isn't any project.

[log.zip](https://github.com/user-attachments/files/19369931/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

Noisyfox, 2025-03-22:
Did your printer run in lan mode?

DiomedesDominguez, 2025-03-22:
Yes, the printer is running in LAN mode. With the RC version I haven't seen
any problem.

Hansi-Sims, 2025-08-27:
I'm experiencing the same issue with my setup. I'm using Linux Mint 22.1 with an Nvidia RTX 2080 graphics card. The printer is an Elegoo Neptune 4 with Moonraker/Fluid/Klipper firmware. Orca works fine initially, but every time I try to connect via the Device tab, the program becomes unresponsive and crashes.

Are there any solutions to this problem?