OrcaSlicer/OrcaSlicer #8982: Lost all printers, filaments and printer profiles

thedizze, 2025-03-20:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

windows 10

### Printer

custom printers profiles twotrees, vzbot, and geeetech A10

### How to reproduce

I've just installed new 2.3.0 version and uninstalled 2.3.0-rc by prompt request during installation and all my settings were lost. 
all my printers profiles are gone all my filaments and also print profiles.
they were 4 printers about 100 filaments and per printer differente quality settings for every printer

my \AppData\Roaming\OrcaSlicer\user\default folders are all empty and I don't have any saved backup

are all my settings gone forever?

### Actual results

lost all saved presets and settings

### Expected results

after the update all saved settings should be automatically restored 

### Project file & Debug log uploads

[log.zip](https://github.com/user-attachments/files/19371186/log.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

Setup wizard doesn't work and open a blank windows

SoftFever, 2025-03-21:
Hi 
Orca should backup your user folder. Can you find folder like AppData\Roaming\OrcaSlicer\user-backup-v2.3.0?

thedizze, 2025-03-21:
hi yes I found that folder but it is empty
I've also tried to revert back on a previous version and uninstall and
reinstall orcaslicer on previous stable version but nothing.
seem that all my printer profiles, filament and settings are lost. I'm bit
upset since they were 3 custom printers with klipper and about 100
filaments.
also if I try to add or create a new printer just a white window is showing
(see attachment).

Your help is appreciated

thedizze, 2025-03-21:
now I'm back on 2.2.0 stable version

SoftFever, 2025-03-22:
> hi yes I found that folder but it is empty
> I've also tried to revert back on a previous version and uninstall and
> reinstall orcaslicer on previous stable version but nothing.
> seem that all my [...]

This is unusual; it suggests that the folder was emptied by another software.
Did you previously specify the “data_dir” option for OrcaSlicer? If that's the case, your privious OrcaSlicer were not using \AppData\Roaming\OrcaSlicer folder to store custom profiles.
Another possibility is that you have a “deep uninstall” application installed on your system that automatically cleans up “remainders” when uninstalling a software.
OrcaSlicer’s uninstaller won’t delete custom profiles.

thedizze, 2025-03-22:
at this point you make me think that could be iobit uninstaller the culprit

SoftFever, 2025-03-22:
> at this point you make me think that could be iobit uninstaller the culprit
> 
> Il sab 22 mar 2025, 02:47 SoftFever ***@***.***> ha scritto:
> […](#)

That is it.
It deleted OrcaSlicer's associated roaming folder where custom settings are stored.

BiomedicLJZ, 2025-03-23:
I also lost all my printers, settings and filaments with this version but as soon as i rolled back to 2.2.0 **everything came back**, i also wasn't able  to create a profile for me **Ender 3v3 KE**

matikij, 2025-03-23:
I think I have a related issue on Linux. I am unable to see some of my custom filament profiles in **2.3.0**, but when I start the 2.2.0 AppImage I can see all of them.
Tried looking at the missing profiles and there's nothing standing out. Attached the .json from one of them to this message

The contents of `.info` are

```
sync_info = create
user_id = 
setting_id = 
base_id = GFSL99
updated_time = 1691700349
```

[eSUN silk PLA.json](https://github.com/user-attachments/files/19411226/eSUN.silk.PLA.json)

thedizze, 2025-03-27:
I didn't solve it I have to redo all my printers and filaments from zero even with recuva or others softwares I was not able to recover lost files

closed 2025-03-27 by thedizze

Dygresja666, 2025-03-29:
Hi there, 
I also lost my custom filaments and printers after upgrading to 2.3.0 from 2.3.0-rc. I do not use any uninstaller software on my W11.  I was recreating the profiles, and today, they've disappeared once again. 
Uninstalled the 2.3.0 and installed the 2.2.) my lost profiles (the previous ones from over a week appeared in the slicer!

thedizze, 2025-03-29:
@Dygresja666 lucky you after wasting an entire week trying to retrieve them I've started today to rebuild all my profiles

Jax1er, 2026-01-07:
All my custom filament profiles are also gone 2.3.1.  I can see them in the AppData\Roaming\OrcaSlicer\user-backup-v2.3.0
the filament I am seeing looks like its from AppData\Roaming\OrcaSlicer\system\OrcaFilamentLibrary\filament.  I have tried to paste the backup filament into this folder and other folders labeled filament but no custom filaments.

derrickwzb, 2026-02-23:
@Jax1er If the backup from your "AppData\Roaming\OrcaSlicer\user-backup-v2.3.0\default\filament\base" is correct and has the filaments that are missing, you should copy it towards "AppData\Roaming\OrcaSlicer\user\default\filament\base" and it should reflect in Orcaslicer.