OrcaSlicer/OrcaSlicer #8988: Supports are not auto generating as before in 22.

Phardbean, 2025-03-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

23 stable, first occurred in 23 Rc, but not beta

### Operating System (OS)

Windows

### OS Version

11 home 12.0.26100

### Additional system information

OS Name	Microsoft Windows 11 Home
Version	10.0.26100 Build 26100
Other OS Description 	Not Available
OS Manufacturer	Microsoft Corporation
System Name	ARCLIGHT
System Manufacturer	Micro-Star International Co., Ltd.
System Model	MS-7E12
System Type	x64-based PC
System SKU	To be filled by O.E.M.
Processor	AMD Ryzen 9 7950X3D 16-Core Processor, 4201 Mhz, 16 Core(s), 32 Logical Processor(s)
BIOS Version/Date	American Megatrends International, LLC. 1.E0, 7/26/2024
SMBIOS Version	3.7
Embedded Controller Version	255.255
BIOS Mode	UEFI
BaseBoard Manufacturer	Micro-Star International Co., Ltd.
BaseBoard Product	MAG X670E TOMAHAWK WIFI (MS-7E12)
BaseBoard Version	1.0
Platform Role	Desktop
Secure Boot State	On
PCR7 Configuration	Elevation Required to View
Windows Directory	C:\WINDOWS
System Directory	C:\WINDOWS\system32
Boot Device	\Device\HarddiskVolume1
Locale	Germany
Hardware Abstraction Layer	Version = "10.0.26100.1"
User Name	
Time Zone	W. Europe Standard Time
Installed Physical Memory (RAM)	64.0 GB
Total Physical Memory	63.1 GB
Available Physical Memory	49.8 GB
Total Virtual Memory	67.1 GB
Available Virtual Memory	50.2 GB
Page File Space	4.00 GB
Page File	C:\pagefile.sys
Kernel DMA Protection	Off
Virtualization-based security	Not enabled
App Control for Business policy	Enforced
App Control for Business user mode policy	Off
Automatic Device Encryption Support	Elevation Required to View
Hyper-V - VM Monitor Mode Extensions	Yes
Hyper-V - Second Level Address Translation Extensions	Yes
Hyper-V - Virtualization Enabled in Firmware	Yes
Hyper-V - Data Execution Protection	Yes
-----------------------------------------------------------------------

### Printer

Bambu P1S

### How to reproduce

..... has something changed in the native bridging or support settings??? I haven't changed anything. It only effects my P1S, not my MK4S HF. a new file does nothing. As I write this, I am going to try with a completely "clean" profile??? i don't know

### Actual results

When I use my MK4S profile, in generates support on the same exact file.

### Expected results

https://youtu.be/NmjPNgm6hLY    Support??? ha!

### Project file & Debug log uploads

[OrcaFileZip.zip](https://github.com/user-attachments/files/19386269/OrcaFileZip.zip)
[OrcaLogZip.zip](https://github.com/user-attachments/files/19386268/OrcaLogZip.zip)

you know, a 3mf file is a zip??? why make it so complicated to unzip then rezip?

### Checklist of files to include

- [x] Log file
- [x] Project file

### Anything else?

42

Phardbean, 2025-03-21:
I should clarify; I use the same exact support settings. as far as I know ....

Phardbean, 2025-03-21:
Reverted to beta, everything fine.

Phardbean, 2025-04-03:
Okay so like ... this is still in issue for me ...

faberc, 2025-04-06:
Adding to say I am experiencing the same on 2.3.0 Build 70931e5. Have to manually paint on supports to get any supports to generate when slicing.

Phardbean, 2025-04-12:
I think it's time for me to move away from orca.

closed 2025-07-19 by github-actions[bot]