OrcaSlicer/OrcaSlicer #8990: flushing volumes per filament /filament type change

avxmw, 2025-03-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Just had a part breaking as there was not enough flushing of filaments when changing from PETG to PLA support and back. While the part printed successully, it broke immediately while trying to actually make use of it.

So far only color is taking into account with regards to flushing volumes, material changes should also be taken into consideration and with a higher priority.

Essentially stale and closed bug Stale bug https://github.com/SoftFever/OrcaSlicer/issues/4783 essentially.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

Add a way to specify flushing volums per material (change) or a fixed multiplier.

### Describe alternatives you've considered

1. not use different material as support if that leads to increased risk of the part becoming unusable

slothy89, 2025-03-22:
Having a material swap multiplier would be a good option. You could set your flush volumes based on colour, then have a second multiplier for inter-material, then you could set it to say 3x for PLA>PETG etc.

Currently you can just manually edit the flush volume values when you know this is going to be the case. I've found 500-600mm3 a good minimum for PETG to PLA. 800mm3 to be safe.

I have noticed that using Bambu's Support for PLA it already auto sets the flush to a fixed value across all colours. in this example Filament 4 is the Support for PLA.

[image]

pravipoldi, 2025-04-11:
Flushing volumes for filament change for P1S constantly goes to 0,00. I must set it every time before slicing, no mather what model.

Le-zOU, 2025-05-29:
try this

[image]