OrcaSlicer/OrcaSlicer #8991: Is the newest version malware?

squidwardissquidward, 2025-03-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 11

### Printer

No printer because this is not an issue with app

### How to reproduce

1. Download the file
2. Go to [virustotal.com ](https://www.virustotal.com/)
3. Upload the file

### Actual results

[image]

### Expected results

It should be this but it should be be blue or green instead of red.

[image]

### Project file & Debug log uploads

I was using the newest version( not portable) but for some reason I cant upload it.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Anything else?

[image]

MxBrnr, 2025-03-21:
If you downloaded the file from GitHub or the official OrcaSlicer website (www.orcaslicer.com), then no, it is not malware; you are probably just getting a false positive.

But if you downloaded the file from anywhere else then the security of the file cannot be trusted.

Zogar89, 2025-03-21:
Windows says this file is not secure.

MxBrnr, 2025-03-21:
> Windows says this file is not secure.

This is also a false positive. It's only because the Orca devs have not bothered with the "certificates" that Windows requires to NOT flag it. The program is perfectly safe if it is downloaded from the official sources.

Bourbon697, 2025-03-23:
I am having the same issue and cant install

[image]

MxBrnr, 2025-03-23:
> I am having the same issue and cant install or even open this

You can if you know where the option is in whatever software you are using.

closed 2025-06-29 by github-actions[bot]