OrcaSlicer/OrcaSlicer #9002: Exhaust fan incorrectly turned on by default in a lot of filament profiles

Snelso91, 2025-03-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

Windows 10 64 bit

### Printer

Bambu Lab P1S

### How to reproduce

Check the filament profiles for some of the Bambu technical filaments such as Bambu ABS, ASA, PAHT-CF, PC, etc.
I've seen some information that suggests that this issue is affecting filament profiles from other manufacturers as well, for example the "Voron Generic ABS" profile as shown in #8572.

### Actual results

These filaments all have the "activate air filtration" checkbox enabled, and the "during print" and "complete print" percentages set to the default of 70%. Therefore, the exhaust fan is turned on during the print and remains on after the print.

### Expected results

These filaments should instead have the "activate air filtration" checkbox disabled by default in order to keep the exhaust fan off both during and after the print.
This is because they are all filaments that are best printed with a warm/hot chamber, and so any chamber cooling due to the exhaust fan pulling in fresh air into the chamber (due to the negative pressure the fan creates) can cause warping or print failures.
If you look at Bambu Studio for example and slice a generic cube (such as in the 3mf I have attached) with one of the mentioned Bambu filaments selected, then in the output gcode there are no commands to turn on the exhaust fan, which is what I would expect.
Whereas in Orca, the exhaust fan is incorrectly turned on for all of these filaments.

### Project file & Debug log uploads

[Exhaust fan bug.zip](https://github.com/user-attachments/files/19397489/Exhaust.fan.bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

### Anything else?

[image]

GoMachV, 2025-04-13:
I am having this same issue. I am a bit confused though, its been a while since reported and this is the only report I see. Do I need to do anything to echo I am having the same issue? I know there are workarounds but every time I sync my material I have to then remember to go back and change its profile.

Snelso91, 2025-04-21:
> I am having this same issue. I am a bit confused though, its been a while since reported and this is the only report I see. Do I need to do anything to echo I am having the same issue?

@GoMachV I'm by no means an expert, but this is my understanding of how things work here based on the bug reports I've submitted here in the past:

Generally, unless you are able to program a fix for the bug and submit it to this github as a pull request, then the most we can do is just file one bug report (like I have done here) and then other people such as yourself can leave messages on that bug report discussing the bug or saying if they are also experiencing it.
Every 3 months or so the github bot will warn you that it is going to close the bug report because it has been inactive for too long, and so in that case all you need to do is check if the bug is still present in the latest version of Orca, and if so then simply leave a comment on the bug report saying that it is still not fixed, and this should prevent the bot from closing the issue unnecessarily.

But unfortunately other than those things, there is not much you or I can do to get the bug fixed because like I said unless you are a decent programmer and also you know how the Orca Slicer code works, then instead you will have to wait for the developers to see this bug report and then maybe they might fix it or they might not.
As far as I know, OrcaSlicer is open source and so is maintained by volunteers, therefore there is no guarantee that bugs will be fixed quickly or at all, since there are no salaried permanent "staff", and so fixing bugs and adding new features are only done if people volunteer their time for free to do those things.
I've reported a few bugs for Orca Slicer so far, all of them I have followed the bug report template and tried to be as detailed as possible about the nature of the bug, as well as attaching log and project files where needed. My oldest Orca bug report was created in January 2024 and it still hasn't been fixed and got automatically closed by the github bot despite still being present in the latest Orca version.
In addition, you can see that there are approx. 1000 open issues on this github, therefore my conclusion is that there are simply too many bugs in the software and not enough volunteers who are able to fix them. And so it is likely that many of these bugs will either never be fixed, or will take years to be fixed. Or it's possible it could be fixed within days of the bug report being filed, but based solely on my experience, that almost never happens unfortunately.

Snelso91, 2025-07-21:
This bug is still present in the latest release (v2.3.0).

keredg, 2025-10-18:
In the latest version V2.3.1 after a print with ABS the Exhaust Fan continues at 70% for ever if left.
In the Bambu Lab ABS filament profile in cooling the Exhaust fan both during print and on completion are both set to 70% so the fan never stops until you switch the printer off or start a new print with other material

AGluk, 2026-11-08:
I can confirm that this issue also affects the **Bambu Lab H2D running OrcaSlicer 2.5.0**.

When printing engineering materials such as **ABS, ASA, PA, and PPA**, OrcaSlicer generates G-code that activates the Chamber Exhaust Fan during printing and keeps it running after printing.

This noticeably increases printer noise and may interfere with maintaining a stable chamber temperature.

**Bambu Studio behaves differently:** using comparable printer and filament settings, it does not generate the additional Exhaust Fan commands, even when `activate_air_filtration = true` is present in the filament profile.

There is also an important distinction in the H2 series:

- The H2D has separate Heating and Cooling airflow modes.
- In Heating mode, external airflow shutters are closed, and internal chamber circulation is handled automatically.
- The printer's actual filtration behavior is controlled separately from the legacy `activate_air_filtration` setting.
- Bambu Studio no longer exposes this legacy Air Filtration Enhancement setting for H2-series printers, whereas OrcaSlicer still does.

This suggests that OrcaSlicer may be applying legacy Exhaust Fan control logic to printers whose airflow management has changed.

Interestingly, the Exhaust Fan settings were explicitly restored for the H2D in **#13384**, following **#13378**. However, exposing these settings does not necessarily mean that enabling the fan automatically is appropriate for this printer.

**Suggested solution:**

Please review the default Exhaust Fan behavior for the H2 series and consider aligning it with current Bambu Studio behavior.

Ideally, OrcaSlicer should not automatically activate the Exhaust Fan for high-temperature materials unless explicitly requested by the user.

Manual Exhaust Fan control should remain available for those who need it.

This would reduce unnecessary noise and avoid potentially undesirable chamber cooling during engineering-material printing.

Thank you!