OrcaSlicer/OrcaSlicer #9003: Variable Layer height Adaptive Button/Function not working

Sledgehammer88, 2025-03-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0

### Operating System (OS)

Windows

### OS Version

11

### Additional system information

on several systems with AMD/Intel CPU; 16-32GB RAM; integrated and external GPU

### Printer

Bambu Lab P1S / X1C

### How to reproduce

1. Select an object
2. Click on "Variable Layer Height"
3. Adjust "Quality  / Speed"
4. Click on the "Adaptive" Button

### Actual results

Layer heights will not be adjusted. If they were adjusted before in V2.2, the heights will be reset to stock.

### Expected results

Adjusting the layer height pending to the "Quality  / Speed" setting

### Project file & Debug log uploads

[debug_Wed_Mar_19_17_35_37_23700.log.zip](https://github.com/user-attachments/files/19398243/debug_Wed_Mar_19_17_35_37_23700.log.zip)

[Planetary+Gear+Spinner+Light2.zip](https://github.com/user-attachments/files/19398303/Planetary%2BGear%2BSpinner%2BLight2.zip)

### Checklist of files to include

- [x] Log file
- [ ] Project file

### Anything else?

If the Project is opened in v2.2.0 again, it works correctly.

Evixone, 2025-03-22:
I confirm, orca 2.3.0, adaptive layer heights no longer work.

BodyBas, 2025-05-10:
Still no adaptive layer height in version 2.3.0 Build_70931e5.

closed 2025-08-17 by github-actions[bot]