OrcaSlicer/OrcaSlicer #9004: Orca Slicer Major Opportunity

10gregoryc, 2025-03-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

With all traditional printers, they have a fixed size on the X, Y, and Z axis. The only type of printer that does not have this constraint are belt printers. They have a constraint on size on the X and Y axis, but they have an unlimited size on the Z. This sounds great. This allows a small printer to be far more versatile. However when I look at them they have only one flaw. They don’t have the support in the software. They have very few options for slicers, and the ones available are not as smooth of an experience as orca slicer. People have to use slicers like the clunky Ideamaker, or Creality slicer, which no longer supports belt printers. Belt printers can be really good. Printers like the Ideaformer IR3 V2 show this, and there is a whole area of printing hobbyists who want to be able to use their printer with orca slicer, but are unable to because no good slicer company provides them with a way to slice their models quickly and efficiently.

### Which printers will be beneficial to this feature?

All

### Describe the solution you'd like

If you were to add support for this type of printer in the way of profiles or even the option to make your own profiles for belt printers, many of those people will turn to orca slicer as their default. I may be overestimating the number of people who want this, but it still is something that I am sure many people would appreciate. I know I would appreciate it.

OptikCrash, 2025-05-04:
I would love to see this as well.  Maybe [@Raise3D](https://github.com/Raised3D) should contribute to a branch?

tommasobbianchi, 2025-05-22:
I would really love to see this implementation

closed 2026-10-09 by SoftFever