OrcaSlicer/OrcaSlicer #9005: Flashforge webui ifs connect

TMTYD, 2025-03-21:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Unfortunately, the web interface does not work for the ad5x, so you cannot select the filament using IFS

[image]
[image]

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

It would be good if the connection to the printer worked, then you could select the right color in the IFS

TMTYD, 2025-03-21:
This is how it is in the orca from flashforge

[image]

TMTYD, 2025-03-30:
Can this be fixed?

GhostTypes, 2025-07-10:
I've started initial work on adding support for the Adventurer 5X to my own projects and will provide more details as they become available.

## Overview

The AD5X maintains the same communication structure of the Adventurer 5 series, with modifications centered around the Material Station.

I've identified the following changes so far:

  * **Removed Features**:
      * `M661` / `M662`: These legacy commands for requesting the full local file list and individual file PNG data have been removed. This is likely because the AD5X is not supported in FlashPrint, which relied on this API.
      * All other G/M-code commands functional on the Adventurer 5M/Pro are expected to work on the AD5X. I have tested the 'basics' like M601/M602, M114, M115, M119, etc.

-----

### AD5X File Upload Request

It does not appear that you can upload a file *without* it automatically starting.

```json
{
  "method": "POST",
  "endpoint": "/uploadGcode",
  "protocol": "HTTP/1.1",
  "headers": {
    "Host": "192.168.0.204:8898",
    "Accept": "*/*",
    "serialNumber": "SNMQRE9400951",
    "checkCode": "0e35a229",
    "fileSize": "735095",
    "printNow": "true",
    "levelingBeforePrint": "true",
    "flowCalibration": "false",
    "firstLayerInspection": "false",
    "timeLapseVideo": "false",
    "useMatlStation": "true",
    "gcodeToolCnt": "4",
    "materialMappings": "W3sidG9vbElkIjowLCJzbG90SWQiOjEsIm1hdGVyaWFsTmFtZSI6IlBMQSIsInRvb2xNYXRlcmlhbENvbG9yIjoiI0ZGRkZGRiIsInNsb3RNYXRlcmlhbENvbG9yIjoiI0ZGRkZGRiJ9LHsidG9vbElkIjoxLCJzbG90SWQiOjIsIm1hdGVyaWFsTmFtZSI6IlNJTEsiLCJ0b29sTWF0ZXJpYWxDb2xvciI6IiNGRkZGODAiLCJzbG90TWF0ZXJpYWxDb2xvciI6IiNGRUYwNDMifSx7InRvb2xJZCI6Miwic2xvdElkIjozLCJtYXRlcmlhbE5hbWUiOiJTSUxLIiwidG9vbE1hdGVyaWFsQ29sb3IiOiIjQzBDMEMwIiwic2xvdE1hdGVyaWFsQ29sb3IiOiIjODk4OTg5In0seyJ0b29sSWQiOjMsInNsb3RJZCI6NCwibWF0ZXJpYWxOYW1lIjoiU0lMSyIsInRvb2xNYXRlcmlhbENvbG9yIjoiI0ZGODBDMCIsInNsb3RNYXRlcmlhbENvbG9yIjoiI0YzMzBGOSJ9XQ==",
    "Content-Length": "735315",
    "Content-Type": "multipart/form-data; boundary=--------------------------i7Uk5X8yi9FlS62AcD86Pz"
  },
  "body": {}
}
```
> The body contains raw binary data for the 3mf (or .gcode - need to test) being uploaded.

-----

### Material Mapping Structure (on file upload)

The `materialMappings` parameter is a base64 encoded.. Below is the decoded structure from a print started with Orca-FlashForge. I'm not sure yet if the `toolMaterialColor` and `slotMaterialColor` must match the values reported by the printer's `/detail` endpoint.

```json
"decoded_material_mappings": [
    {
      "toolId": 0,
      "slotId": 1,
      "materialName": "PLA",
      "toolMaterialColor": "#FFFFFF",
      "slotMaterialColor": "#FFFFFF"
    },
    {
      "toolId": 1,
      "slotId": 2,
      "materialName": "SILK",
      "toolMaterialColor": "#FFF80A",
      "slotMaterialColor": "#FEF043"
    },
    {
      "toolId": 2,
      "slotId": 3,
      "materialName": "SILK",
      "toolMaterialColor": "#C0C0C0",
      "slotMaterialColor": "#898989"
    },
    {
      "toolId": 3,
      "slotId": 4,
      "materialName": "SILK",
      "toolMaterialColor": "#FF80C0",
      "slotMaterialColor": "#F330F9"
    }
  ]
```

-----

### AD5X Material Station Info (`/detail`)

The `/detail` endpoint response now includes a `matlStationInfo` object, which provides detailed information on slot configurations and states.

```json
{
  "matlStationInfo": {
    "currentLoadSlot": 0,
    "currentSlot": 0,
    "slotCnt": 4,
    "slotInfos": [
      {
        "hasFilament": true,
        "materialColor": "#FFFFFF",
        "materialName": "PLA",
        "slotId": 1
      },
      {
        "hasFilament": true,
        "materialColor": "#2750E0",
        "materialName": "PLA",
        "slotId": 2
      },
      {
        "hasFilament": true,
        "materialColor": "#FEF043",
        "materialName": "PLA",
        "slotId": 3
      },
      {
        "hasFilament": true,
        "materialColor": "#F95D73",
        "materialName": "PLA",
        "slotId": 4
      }
    ],
    "stateAction": 0,
    "stateStep": 0
  }
}
```

This information should be sufficient for implementing initial support. I will report any further discoveries.

TMTYD, 2025-07-27:
OK, that sounds good, right? Will there be an implementation of IFS Multicolor in the Orca Slicer?