OrcaSlicer/OrcaSlicer #9006: 3MF files with multiple plates result in object position shift when opening

slothy89, 2025-03-21:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

From at least 2.0 to 2.3.0 (latest)

### Operating System (OS)

Windows

### OS Version

Windows 10 - latest

### Printer

Bambu P1S

### How to reproduce

Unsure of exact steps to reproduce, as it doesn't happen 100% of the time. Although it seems to only occur after restarting the computer (cache related?)

1. - Have a project with enough plates to split across 2+ rows, 3+ is more apparent.
2. - Place object on the plates and position as reqiured (eg centred)
3. - Save project
4. - Come back to it at a later time, and re-open the project.
5. - When opening, subsequent rows of plates will have all moved down in the Y-axis, increasing with each row.
6. - Have enough plates, and objects start hanging outside the printable area.

I have seen this happen with pretty much any project that meets the criteria in step 1.

Previously thought it was an issue importing Bambu Studio 3MFs, but it happens to original projects I've made from scratch in Orca.

### Actual results

Here is an after image showing the result of the objects being shifted. All plates were saved with the objects centred on their respective plates.

[image]

[image]

### Expected results

Objects should remain in their original position between openings.

### Project file & Debug log uploads

[CaliDragon_Plate - Base Plate.zip](https://github.com/user-attachments/files/19398461/CaliDragon_Plate.-.Base.Plate.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

SoftFever, 2025-03-22:
I can reproduce it.
Looks like a regression caused by #7923

@Noisyfox do you have any ideas?

Noisyfox, 2025-03-22:
This is expected, as I mentioned in the original PR.

bioxz, 2025-03-22:
I can confirm this issue as well. Consequences: 

1. Some objects ending up on plates, while they where placed outside them on purpose before
2. Some objects are being (partly) placed outsides of plates

The first point required manual intervention, as I had to figure out if those objects belonged to that plate before or not. Just rearranging all objects is not guaranteed to keep the plate assignment coming from 2.2.0.

SoftFever, 2025-03-23:
> This is expected, as I mentioned in the original PR. Any project that saved in Orca 1.6.2 ~ 2.2.0 could have object shifted.

Ah yes, I totally forgot that.
Gonna add a message in the release note later

slothy89, 2025-03-23:
> This is expected, as I mentioned in the original PR. Any project that saved in Orca 1.6.2 ~ 2.2.0 could have object shifted.

Should this apply to brand new projects created recently? ie, v2.2.0? Or should they be immune? I only have P1S profiles, so no changes in bed sizes between profiles.

It also seems to happen to makerworld 3mfs, which I expect is due to Bambu studio not having the same fix.

Noisyfox, 2025-03-23:
> It also seems to happen to makerworld 3mfs, which I expect is due to Bambu studio not having the same fix.

It shouldn't because BBS does not have the bug from the very beginning. Unless you saved the project again in Orca.

slothy89, 2025-03-23:
> > It also seems to happen to makerworld 3mfs, which I expect is due to Bambu studio not having the same fix.
> 
> It shouldn't because BBS does not have the bug from the very beginning. Unless you saved the project again in Orca.

Sorry, yes, this is what I meant. They're fine on first open, but "break" after being resaved.

vrga, 2025-03-31:
Sorry, just to report on this one, i have it happening on 2.3.0 just by saving, switching to new project, then opening the project again.

Plate names also get completely lost.

[image]
[image]

closed 2025-07-07 by github-actions[bot]