OrcaSlicer/OrcaSlicer #9024: Part Cooling Fans not turning on after tool change (missing M106 after tool selection)

traxman25, 2025-03-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.1-dev

### Operating System (OS)

Windows

### OS Version

Windows 10

### Printer

Voron 2.4 StealthChanger

### How to reproduce

Use part cooling fans with any settings.

### Actual results

On Multi tool printers the part cooling fan commands (M106) are not being inserted after a tool change. This results in no part cooling while printing. Variable layer time settings and overhang settings are also being ignored and not inserted into the gcode. 

### Expected results

M106 should be inserted after a tool change to activate part cooling fans on the active tool. Also need to fix the variable layer time iterative part cooling amounts and overhang part cooling fan speed settings. 

### Project file & Debug log uploads

[tool changer part cooling fan bug.zip](https://github.com/user-attachments/files/19404077/tool.changer.part.cooling.fan.bug.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

broncosis, 2025-04-03:
Ia m also seeing this in 2.3 on linux mint 

can we get someone to look at this please

broncosis, 2025-05-11:
[tiny spider_0.2mm_PLA_Generic ToolChanger Printer_1h22m.txt](https://github.com/user-attachments/files/20147472/tiny.spider_0.2mm_PLA_Generic.ToolChanger.Printer_1h22m.txt)

here is the Gcode from a recent attempt this make tool changing almost worthless

broncosis, 2025-06-20:
Check your tool changer config mine was something in the fan settings  
 
 
 
 
 And orca was sending M106.2 which is a Bambu command  
 
 
 
 
 
 
 
 
 awabom  left a comment  (SoftFever/OrcaSlicer#9024) 
 
 For me, the M106 command is inserted exactly BEFORE the toolchange, leading to the wrong tool fan spinning up.

closed 2025-09-27 by github-actions[bot]