OrcaSlicer/OrcaSlicer #9028: Mesh Boolean not removing material

3DeezNutz, 2025-03-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3

### Operating System (OS)

Windows

### OS Version

windows 10

### Additional system information

i7700k
16gb ram

### Printer

voron 2.4

### How to reproduce

add any STL. add a negative part. try to mesh Boolean  using difference. Subtract with negative part. 

### Actual results

the part isnt cut with the negative part.

### Expected results

material removed

### Project file & Debug log uploads

[Mesh Boolean.zip](https://github.com/user-attachments/files/19405778/Mesh.Boolean.zip)

### Checklist of files to include

- [ ] Log file
- [ ] Project file

Ataraxia-Mechanica, 2025-03-23:
[image]
I think your installation might be broken? Looks fine on 2.3.0 downloaded from release page

3DeezNutz, 2025-03-23:
It doesn't cut the original part.

slothy89, 2025-03-23:
I have run into this sometimes too, whether its a negative part or regular part you're using as the subtractor, randomly it doesn't perform the difference operation. I'll take note of it next time I encounter it and add a sample here

Ataraxia-Mechanica, 2025-03-23:
> It doesn't cut the original part. I can slice and print it like that fine. What my goal is to modify that part and save that stl with a larger gap. It's not letting me do that.

Pretty sure orca doesn't have that function as it's quite complex to construct the actual mesh. Just use blender for this instead.

slothy89, 2025-03-23:
> Pretty sure orca doesn't have that function as it's quite complex to construct the actual mesh. Just use blender for this instead.

Why are you even here? Have you even used Orca Slicer?
Orca, Bambu Studio and pretty sure Prusa Slicer all have basic mesh Boolean functionality, and have done for some time.

Ataraxia-Mechanica, 2025-03-23:
> Why are you even here? Have you even used Orca Slicer? Orca, Bambu Studio and pretty sure Prusa Slicer all have basic mesh Boolean functionality, and have done for some time.

I don't think you even understand this. This person is asking for exporting modified mesh which no slicer can do. I suspect the boolean operation is done after converting the mesh to layers of vector images which is easier to do compared to modifying the mesh. Why the hostility when you can't even spend a few seconds to understand what I mean?

slothy89, 2025-03-23:
> I don't think you even understand this. This person is asking for exporting modified mesh which no slicer can do. I suspect the boolean operation is done after converting the mesh to layers of [...]

5-part story:

[image]

[image]

[image]

[image]

[image]

Ataraxia-Mechanica, 2025-03-23:
> 5-part story: Literally all the OP wants to do. And sometimes the difference operation bugs.
You know what? There's a reason I even forgot that button was there, despite I used orca slicer pretty frequently in the last year. That boolean operation must have used floating point numbers, which would fail to slice when 2 surfaces on 2 objects are parallel and very close to each other, and the solver guesses wrong which surface should be on top due to floating point precision. Blender has orders of magnitude more devs and none of them figured out how to make a bug-free floating point solver. Unless the devs rewrite the entire implementation with GMP library's bigint like blender, the boolean bug is here to stay. And yes more complex models makes close surfaces more likely to happen.

Farrit1184, 2025-04-13:
> > 5-part story: Literally all the OP wants to do. And sometimes the difference operation bugs.
> 
> You know what? There's a reason I even forgot that button was there, despite I used orca slicer pretty frequently in the last year. That boolean operation must have used floating point numbers, which would fail to slice when 2 surfaces on 2 objects are parallel and very close to each other, and the solver gets wrong on which surface should be on top, due to floating point precision. Blender has orders of magnitude more devs, and none of them figured out how to make a bug-free floating point solver. Unless the devs rewrite the entire implementation with GMP library's bigint like blender, the boolean bug is here to stay. And yes more complex models makes close surfaces more likely to happen.

IDK why you're being so sour about this. It's worked in previous versions, as I've done this before. *all* slicers have it. (I say this as an industry professional. It's literally my job to set up/repair 3D printers of all flavors)

AlexMa2011, 2025-04-27:
I'm also experiencing this issue, and as a temporary alternative, you can use negative parts or other model editing software

Athiril, 2025-06-22:
> IDK why you're being so sour about this. It's worked in previous versions, as I've done this before. _all_ slicers have it. (I say this as an industry professional. It's literally my job to set [...]

Do you remember which previous version works? I might track it down and use it for negative booleans. I want to use plane cut with registration pins.. which doesnt work and hasnt for a long time (used to work) because it uses negative booleans to cut the holes. Blender isnt a good alternative for this as I want the pins generated with tolerance too.

closed 2025-09-28 by github-actions[bot]

Athiril, 2025-09-29:
Why is this closed? Old versions worked correctly.

Gentronicus, 2025-12-15:
Also having this issue on Ubuntu 25.10 with the V2.3.1 Flatpak release.

ryanrapini, 2026-01-13:
Having this issue right now, can provide a reproduction in 1.3.7.3

Solution was to highlight the main assembly body and choose "Fix model" and afterwards the feature mysteriously worked again