OrcaSlicer/OrcaSlicer #9029: Lines printed to/from prime tower in Timelapse "traditional" mode

KertDawg, 2025-03-22:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 and also 2.2.0

### Operating System (OS)

Windows

### OS Version

Windows 11 24H2

### Additional system information

Alienware M18 R1 AMD
32GB RAM
931GB/351GB free C:
3.63TB/1.87TB free D:

### Printer

Bambu Lab A1 mini

### How to reproduce

1. Create a new project
2. Create two primitive cubes with multiple colors.  I used PLA-White, PLA-Pink, both Generic PLA.
3. Set Timelapse type to "Traditional."
4. Slice plate.
5. See lines running from prime tower to model and back.
6. [image]
7. See that lines are "Prime tower" type and not "Travel."
8. Change Timelapse type to "Smooth."
9. See that these extra lines are not present.
10. [image]
11. When I print a file with the extra lines, they seem to come from the prime tower, not to the prime tower.  That is, they go in one direction.

### Actual results

In traditional timelapse mode, multicolor prints draw lines from the prime tower to the model.  In smooth mode, they are not there.

### Expected results

Traditional mode should not have lines from the prime tower to the model.

### Project file & Debug log uploads

[Cube.zip](https://github.com/user-attachments/files/19405914/Cube.zip)

[log.zip](https://github.com/user-attachments/files/19405924/log.zip)

### Checklist of files to include

- [x] Log file
- [x] Project file

Noisyfox, 2025-03-24:
[image]

Weird... It doesn't happen here.

Noisyfox, 2025-03-24:
Could you upload the gcode file as well?

KertDawg, 2025-03-24:
Here it is.

[Cube_PLA_27m37s.zip](https://github.com/user-attachments/files/19428670/Cube_PLA_27m37s.zip)

KertDawg, 2025-03-24:
I tried the same 3mf in Bambu Studio, and it looks as expected with no extra lines.

KertDawg, 2025-03-27:
I just upgraded Bambu Studio to the latest version 2.0.0.95.  It is now making the same lines from the prime tower in wither timelapse mode.  It did not make those lines as of yesterday.  Is Orca code used by Bambu Studio or something?

KertDawg, 2025-03-27:
I just got around to testing Orca Slicer 2.2.0.  This version does not create the lines from the prime tower.  It seems to work as expected with either timelapse setting.  This seems to have been introduced in 2.3.0, at least for me.

Noisyfox, 2025-03-27:
There might be some issue in the change of certain profiles.

closed 2025-07-03 by github-actions[bot]