OrcaSlicer/OrcaSlicer #9035: OrcaSlicer 2.3.0 ballooning file sizes with tree supports and crashing creality klipper printers

Delta-19a, 2025-03-23:
### Is there an existing issue for this problem?

- [x] I have searched the existing issues

### OrcaSlicer Version

2.3.0 build 70931e5

### Operating System (OS)

Windows

### OS Version

Windows 10

### Additional system information

CPU: Ryzen 5 1600x
Ram:16GB
GPU: 1070Ti

### Printer

Creality Hi

### How to reproduce

When using any type of tree supports file sizes will be 2-3 times bigger when compared to creality print. For some tree support types this will cause your printer to crash around the 65-75% mark, before it crashes you will see the printer hang and freeze well printing the supports. 

[image]

### Actual results

OrcaSlicer will balloon the file size of projects using tree supports. for default, organic and, slim this doesnt lead to any issues. but, for strong this will lead to lag, freezing and, hang ups will printing eventually leading to a crash and XS2000 error on creality klipper machines.
https://www.youtube.com/watch?v=WlkY5LuBbKw

### Expected results

orca to export a file regardless of support style used that wont crash my printer. 

### Project file & Debug log uploads

[strong tree support for Creality Hi.zip](https://github.com/user-attachments/files/19407219/strong.tree.support.for.Creality.Hi.zip)

### Checklist of files to include

- [ ] Log file
- [x] Project file

Noisyfox, 2025-03-23:
Could you upload all the gcode files from your screenshot? I'd like to take a look to see what's different.

Delta-19a, 2025-03-23:
im not smart enough to know whats going on but biggest difference i see is about 30,000 extra lines when compared to creality print 6 gcode

[E.zip](https://github.com/user-attachments/files/19408630/E.zip)

gonna try doing a clean install maybe crap has built up over the last year of orca updates

Noisyfox, 2025-03-23:
I believe it's mainly caused by art fitting. Creality slicer might have higher arc fitting tolerence.

Delta-19a, 2025-03-23:
whoa slow down there, so bambu made a change to how they do supports, when orca last did a merge with bambu this got added to orca, and now a fix is being push to orca that should return this to normal and hopefully fix the issue i found?

Noisyfox, 2025-03-23:
Yes, except the fix is also made by BambuLab, which somehow not been merged into Orca.

Delta-19a, 2025-03-23:
im too dumb to compile this so i'll wait for the next nightly build, thanks for all you guys do.

Delta-19a, 2025-03-23:
so i have been able to do some light testing today and, so far the issue is resolved. orca files are now smaller with and without arc fitting enabled and regardless of arc setting will not crash my creality klipper machine. 

with that being said i would and when time allows be carrying out more test over the next 2 days before i mark this as closed.

PR #9038 (2025-04-01): FIX: enable circle fitting of tree support with larger threshold
jira: STUDIO-8588
Change-Id: Ia566fa14f7f786ff905fc3484885634811a6d686 (cherry picked from commit bambulab/BambuStudio@82c72b19bdaa5d1dd3b7e62faa87a8e9b471ac0a)

Fix #9035

Thanks BambuLab!

Now the file size is reduced a lot when arc fitting is enabled:
[image]
closed 2025-03-26 by Delta-19a