OrcaSlicer/OrcaSlicer #9041: [toolchanger] Support for different nozzles sizes and layer heights in the same print

0101falcon, 2025-03-23:
### Is there an existing issue for this feature request?

- [x] I have searched the existing issues

### Is your feature request related to a problem?

Increase print speeds without decreasing structural integrity of the part. 

### Which printers will be beneficial to this feature?

Klipper

### Describe the solution you'd like

When one wants to print parts which need 40% infill, it would be useful to be able to choose one tool over with a bigger nozzle so extrusion width is bigger (say 0.8mm nozzle) and have it print at a bigger layer height. 

### Describe alternatives you've considered

1. Use bigger nozzles and increase flow rate for one of the toolheads (this artificially increases extrusion width) thus increasing infill without doing it in the software.

johnny-bit, 2025-03-25:
There are multiple other usecases, like print details (text) on the bottom using 0.25 (or smaller) while having the rest of the model printed using 0.5. The thicker infill is a good idea too. Other use case would be similar to https://github.com/TengerTechnologies/Smoothificator but using smaller nozzle for the outside for even smoother surface.

0101falcon, 2025-03-26:
> There are multiple other usecases, like print details (text) on the bottom using 0.25 (or smaller) while having the rest of the model printed using 0.5. The thicker infill is a good idea too. Other [...]

Good point actually. It could be used for multipart, like for example benchy two color, but make one of the parts lower layer height. 

For example: 
Say you have a part which has a thread, and you want to print that thread with 0.1mm layer height, but everything else in 0.2mm layer height for speed. (I know adaptive layers exist, but meh)

johnny-bit, 2025-04-28:
I wonder how https://help.prusa3d.com/article/experimental-printing-with-different-nozzle-diameters_821176 is achieved in prusa slicer and what can be ported over.